syssi/esphome-pace-bms #1: Add PACE BMS support

JacquesdeBruyn, 2023-07-14:
Hi, I'm curious if this is possible. I've gotten a battery with a PACE bms. From what I can tell in my limited knowledge on the subject the protocols appear to be fairly similar. This bms seems to be used in Hubble, Greenrich, SOK, Revov and the KATBATT that I now own. These use rs232 on a RJ12 connector. I'll attach the protocol sheet and a picture of bms front.
[72651.11625-PACE-RS232commuciation-protocal.PACE-RS232-TY16S.-20180705.pdf](https://github.com/syssi/esphome-seplos-bms/files/12000343/72651.11625-PACE-RS232commuciation-protocal.PACE-RS232-TY16S.-20180705.pdf)
[image]

JacquesdeBruyn, 2023-07-14:
[image]
If you check this picture the "Battery" there is the seplos bms and "Test" is the one we are testing here. This one showes up as a battery. I'm thinking it's something to do with the device class?

syssi, 2023-07-14:
Yes. The `device_class` is important here.

I've moved the issue into a new project because it's pretty different to the Seplos project as long we implement the Modbus communication here.

syssi, 2023-07-14:
As soon the `device_class: battery` is attached to the `state of charge` sensor it should be represented as battery.

JacquesdeBruyn, 2023-07-14:
Ok, I have that one setup with your project as a reference but we can get back to that at a later stage. I read through your sample code a couple times and think I'll give adding a couple things a go. Might just need a hand calculating cell delta as I don't seem to see the output for it.

syssi, 2023-07-14:
I've added all register addresses (from the protocol descriptions) as comments to the YAML:

https://github.com/syssi/esphome-pace-bms/blob/main/esp8266-example.yaml#L86-L111

Do you like to contribute some sensors? This example configuration could help you:

https://github.com/syssi/esphome-smg-ii/blob/main/esp8266-example.yaml

I can provide some support to decode the protection bitmasks into readable strings and you are on the right track: If you need the delta of the cell voltages we have to implement a template sensor to calculate the values on demand.

JacquesdeBruyn, 2023-07-14:
I've already mostly typed 3 more that are working. Just juggling some work things aswell.

JacquesdeBruyn, 2023-07-14:
So...I got all the sensors I think are worth capturing. Getting a reading of 65535 on all the temperature sensors as well as some of the cells showing voltages like 0.168 and 0.226(I'm going to check these on the bms itself now). Also where do I put the yaml for you?

JacquesdeBruyn, 2023-07-14:
> I've added all register addresses (from the protocol descriptions) as comments to the YAML:
>
> https://github.com/syssi/esphome-pace-bms/blob/main/esp8266-example.yaml#L86-L111
>
> Do you like to [...]

Yes I like to have access to cell delta voltage as this is a good indicator of overall pack health

JacquesdeBruyn, 2023-07-14:
Working so far: SOC, SOH, Current in/out, Total voltage, Remaining capacity in AH, Full and design capacity(same result), cycle count and cell 1-5 voltages.
Not working yet: All temperature sensors and cell 6-16(wildly different results here).

Code so far:

substitutions:
  name: pace-bms-test
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: esphome-web-9d9124
  friendly_name: Test

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "VWCTZnKZwlLn5Z5vNwdvPk0+51EFx2radcwh9g0BbJY="

ota:

logger:
  level: DEBUG

uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: false

modbus:
  - id: modbus0
    flow_control_pin: 18
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

sensor:
  # 0x0000 Current   2 byte   R/INT16  10mA
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} Current"
    address: 0x0000
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "A"
    device_class: current
    state_class: measurement
    accuracy_decimals: 2
    filters:
[... 367 lines cut ...]

  # 0x0035  Mosfet temperature  2 byte   R/INT16   0.1˚C
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} Mosfet temperature"
    address: 0x0035
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "˚C"
    device_class: temperature
    state_class: measurement
    accuracy_decimals: 2
    

  # 0x0036  Environment temperature  2 byte   R/INT16   0.1˚C
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} Environment temperature"
    address: 0x0036
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "˚C"
    device_class: temperature
    state_class: measurement
    accuracy_decimals: 2

syssi, 2023-07-15:
I've merged your definitions into the YAML: https://github.com/syssi/esphome-pace-bms/blob/main/esp8266-example.yaml

And I've change the format of the Modbus registers from Hex to Int. I assume the will fix all `cell voltage` and `battery temperature` issue. Please give it a try!

JacquesdeBruyn, 2023-07-15:
Most of what you changed is gibberish to me, I just know enough about code to be a danger lol.
Anyway, it's uploading

JacquesdeBruyn, 2023-07-15:
Might want to uncomment the battery class from the cycle counter. Other than that it is working 100%

syssi, 2023-07-15:
Why do you want to use the device class here? Does it have any effect or do you just like the icon?

JacquesdeBruyn, 2023-07-15:
I meant remove it, in my code it was still listed as battery which isn't needed.
I also don't know how to go about the rest of the calculations, I haven't found anything on the seplos side I can use and I don't know how to go about calling sensors up for calculations in esphome itself.

syssi, 2023-07-15:
I've removed the device class `battery` now. It's only required/useful for the `state of charge`  sensor. See https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes

syssi, 2023-07-15:
> I also don't know how to go about the rest of the calculations, I haven't found anything on the seplos side I can use and I don't know how to go about calling sensors up for calculations in esphome itself.

I will improve the configuration as soon as possible and add a delta cell voltage sensor.

JacquesdeBruyn, 2023-07-15:
No rush, it's working well enough already. And I'm sure I can add all of those through Hass helpers if I need them. 
Thanks for all the help so far, it's greatly appreciated

syssi, 2023-07-15:
I've added a `power` sensor to the YAML. Please give it a try.

JacquesdeBruyn, 2023-07-15:
Responds as unknown in the sensor list. In the logs it seems to show the battery voltage x100

syssi, 2023-07-15:
I've extended the lambda function with some debug statements. Could you flash the ESP again and provide a log? I would like to understand the structure of the `data` object. I guess I've to rearrange the data types (uint16_t vs. unt8_t).

JacquesdeBruyn, 2023-07-15:
Log response for that section:
[13:02:02][D][modbus_controller.sensor:025]: Sensor new state: nan
[13:02:02][D][sensor:094]: 'pace-bms power': Sending state nan W with 2 decimals of accuracy

syssi, 2023-07-15:
There should be some more log messages because of these lines:

```
      ESP_LOGI("", "Lambda incoming value=%f - data array size is %d", x, data.size());
      ESP_LOGI("", "Sensor properties: address = 0x%X, offset = 0x%X", item->start_address, item->offset);
      int i=0;
      for (auto val : data) {
        ESP_LOGI("", "data[%d]=0x%02X (%d)", i, data[i], data[i]);
        i++;
      }
```

See https://github.com/syssi/esphome-pace-bms/blob/main/esp8266-example.yaml#L92-L117

JacquesdeBruyn, 2023-07-15:
[13:14:02][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[13:14:02][D][modbus_controller.sensor:025]: Sensor new state: 3674.00
[13:14:02][D][sensor:094]: 'pace-bms current': Sending state 36.74000 A with 2 decimals of accuracy
[13:14:02][I][:113]: Lambda incoming value=3674.000000 - data array size is 16
[13:14:02][I][:114]: Sensor properties: address = 0x0, offset = 0x0 value type=1
[13:14:02][I][:117]: data[0]=0x0E (14)
[13:14:02][I][:117]: data[1]=0x5A (90)
[13:14:02][I][:117]: data[2]=0x15 (21)
[13:14:02][I][:117]: data[3]=0x50 (80)
[13:14:02][I][:117]: data[4]=0x00 (0)
[13:14:02][I][:117]: data[5]=0x54 (84)
[13:14:02][I][:117]: data[6]=0x00 (0)
[13:14:02][I][:117]: data[7]=0x64 (100)
[13:14:02][I][:117]: data[8]=0x29 (41)
[13:14:02][I][:117]: data[9]=0x34 (52)
[13:14:02][I][:117]: data[10]=0x30 (48)
[13:14:02][I][:117]: data[11]=0xD4 (212)
[13:14:02][I][:117]: data[12]=0x30 (48)
[13:14:02][I][:117]: data[13]=0xD4 (212)
[13:14:02][I][:117]: data[14]=0x00 (0)
[13:14:02][I][:117]: data[15]=0x06 (6)
[13:14:02][D][modbus_controller.sensor:025]: Sensor new state: nan
[13:14:02][D][sensor:094]: 'pace-bms power': Sending state nan W with 2 decimals of accuracy
[13:14:02][D][modbus_controller.sensor:025]: Sensor new state: 5456.00
[13:14:02][D][sensor:094]: 'pace-bms total voltage': Sending state 54.56000 V with 2 decimals of accuracy
[13:14:02][D][modbus_controller.sensor:025]: Sensor new state: 84.00

syssi, 2023-07-15:
Perfect! I know what's going on and will provide a fix.

syssi, 2023-07-15:
I've pushed a fix: https://github.com/syssi/esphome-pace-bms/blob/main/esp8266-example.yaml#L92-L112

Please give it another try! :-)

JacquesdeBruyn, 2023-07-15:
It's working, I'll figure out how much the value needs to be adjusted and let you know

syssi, 2023-07-15:
Fixed: `      return current * total_voltage * 0.0001f;`

syssi, 2023-07-15:
I will do something similar to calculate the cell delta voltage later the day. I will ping you as soon it's ready.

JacquesdeBruyn, 2023-07-15:
Sure, I'll quickly verify this again. I have a birthday function this evening, just incase you miss me.

syssi, 2023-07-15:
I've added some more sensors (min/max cell voltage, average and delta cell voltage). Please give it a try!

syssi, 2023-07-15:
@FCLFCL82 Could you try the improved YAML configuration too? Could you provide some details about your battery pack and BMS? I would like to extend the list of supported devices.

JacquesdeBruyn, 2023-07-15:
> I've added some more sensors (min/max cell voltage, average and delta cell voltage). Please give it a try!

I'll test when I'm back home👍🏻

FCLFCL82, 2023-07-15:
> @FCLFCL82 Could you try the improved YAML configuration too? Could you provide some details about your battery pack and BMS? I would like to extend the list of supported devices.

That would be great. I tried the improved YAML but didnt get any results, log below.  I got some readings with your initial code.  

My BMS is a Gobel Power RN150, the manufacturer sent me this link for modbus details: [](https://github.com/fancyui/Gobel-Power-RN-BMS-RS485-ModBus)

[00:19:19][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[00:19:19][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[00:19:20][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[00:19:20][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[00:19:20][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[00:19:20][D][modbus_controller:029]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[00:19:20][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[00:19:20][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[00:19:21][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[00:19:21][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[00:19:21][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[00:19:21][D][modbus_controller:029]: Modbus command to device=1 register=0x09 countdown=0 no response received - removed from send queue
[00:19:21][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[00:19:22][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[00:19:22][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[00:19:22][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[00:19:22][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[00:19:22][D][modbus_controller:029]: Modbus command to device=1 register=0x0F countdown=0 no response received - removed from send queue
[00:19:22][D][uart_debug:114]: >>> 01:03:00:69:00:02:14:17
[00:19:23][D][uart_debug:114]: >>> 01:03:00:69:00:02:14:17
[00:19:23][D][uart_debug:114]: >>> 01:03:00:69:00:02:14:17
[00:19:23][D][uart_debug:114]: >>> 01:03:00:69:00:02:14:17
[00:19:23][D][uart_debug:114]: >>> 01:03:00:69:00:02:14:17
[00:19:23][D][modbus_controller:029]: Modbus command to device=1 register=0x69 countdown=0 no response received - removed from send queue

syssi, 2023-07-16:
@FCLFCL82 Do you get a response if you remove all sensors expect the `total voltage` and `state of charge` again? Did you mess up your wiring in the meantime or did you use the wrong RS485 port? Did you change the dip switch configuration?

JacquesdeBruyn, 2023-07-16:
Works well, I just adjusted the delta voltage to 3 decimal places. It's a fairly small value.

syssi, 2023-07-16:
@JacquesdeBruyn Could you provide some details about your battery pack and BMS model? So I could add your setup to the list of supported devices already.

JacquesdeBruyn, 2023-07-16:
What details do you need?

syssi, 2023-07-16:
> The BMS is a Gobel Power RN150, the manufacturer sent me this link for modbus details: [Gobel Power RN BMS](https://github.com/fancyui/Gobel-Power-RN-BMS-RS485-ModBus)

This repository describes a different (ASCII based / similar to the Seplos) protocol and no Modbus RTU. The project name is misleading unfortunately. You proved already your BMS responds to Modbus RTU too.

syssi, 2023-07-16:
> Version number: P16S200A-21473-1.03A310771030800025A

Perfect! Is this the string reported by the `version number` sensor?

JacquesdeBruyn, 2023-07-16:
Yes it is

JacquesdeBruyn, 2023-07-16:
Anything else you want tested? 
Before I try combining both onto one esp and skuff everything up properly :D

syssi, 2023-07-16:
Do you like to test writing into registers? Please keep in mind it could be dangerous.

syssi, 2023-07-16:
This is your battery bank, right? https://katbatt.co.za/products/6-4kwh-lifepo4-battery

JacquesdeBruyn, 2023-07-16:
I'm not sure if there are registers I can mess with without affecting the warranty. I'll go through them in a bit. I'd actually like to adjust the low cutoff. My inverter isn't "smart" so I manage things around it. It's a beast of an inverter, never had any issues with it. Apart from the low cutoff not being able to adjust high enough.

syssi, 2023-07-16:
I've implemented two `number` entities to adjust some balancer settings. Changing these settings shouldn't shutdown the BMS accidentally. Does these entities show proper values (in `mV` at the moment). Are you able to adjust the setting using the number entity? Does the BMS accept the new settings or does the number entity gets resetted to the initial value?

JacquesdeBruyn, 2023-07-16:
I'm not at home right now. I'll check those in an hour or so

syssi, 2023-07-16:
Take your time! :-)

JacquesdeBruyn, 2023-07-16:
INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/esphome-web-9d9124.yaml...
Failed config

sensor.modbus_controller: [source /config/esphome/esphome-web-9d9124.yaml:638]
  platform: modbus_controller
  modbus_controller_id: bms0
  name: pace-bms balance start cell voltage
  
  [use_write_multiple] is an invalid option for [sensor.modbus_controller]. Please check the indentation.
  use_write_multiple: True
  address: 105
  register_type: holding
  value_type: U_WORD
  
  [min_value] is an invalid option for [sensor.modbus_controller]. Did you mean [on_value], [on_raw_value], [on_value_range]?
  min_value: 0.0
  
  [max_value] is an invalid option for [sensor.modbus_controller]. Did you mean [on_value], [on_raw_value]?
  max_value: 255.0
  
  [step] is an invalid option for [sensor.modbus_controller]. Please check the indentation.
  step: 1
  unit_of_measurement: mV
sensor.modbus_controller: [source /config/esphome/esphome-web-9d9124.yaml:651]
  platform: modbus_controller
  modbus_controller_id: bms0
  name: pace-bms balance start delta voltage
  
  [use_write_multiple] is an invalid option for [sensor.modbus_controller]. Please check the indentation.
  use_write_multiple: True
  address: 106
  register_type: holding
  value_type: U_WORD
  
  [min_value] is an invalid option for [sensor.modbus_controller]. Did you mean [on_value], [on_raw_value], [on_value_range]?
  min_value: 0.0
  
  [max_value] is an invalid option for [sensor.modbus_controller]. Did you mean [on_value], [on_raw_value]?
  max_value: 255.0
  
  [step] is an invalid option for [sensor.modbus_controller]. Please check the indentation.
  step: 1
  unit_of_measurement: mV

I'm removing the setting parts for now and just getting the response values

JacquesdeBruyn, 2023-07-16:
Balance start is set at 3.4v and the delta at 20mv

syssi, 2023-07-16:
Did you mess up your YAML? It's important to add the new entities to the `number:` section:

```
number:
  # 105  Balance start cell voltage            2 byte  RW  uint16  mV
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} balance start cell voltage"
    use_write_multiple: true
    address: 105
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 65535.0
    step: 1
    unit_of_measurement: "mV"

  # 106  Balance start delta voltage           2 byte  RW  uint16  mV
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} balance start delta voltage"
    use_write_multiple: true
    address: 106
    register_type: holding
    value_type: U_WORD
    min_value: 0.0
    max_value: 65535.0
    step: 1
    unit_of_measurement: "mV"
```

FCLFCL82, 2023-07-16:
> @FCLFCL82 Do you get a response if you remove all sensors expect the `total voltage` and `state of charge` again? Did you mess up your wiring in the meantime or did you use the wrong RS485 port? Did you change the dip switch configuration?

Something is not right. I'll review everything again

JacquesdeBruyn, 2023-07-16:
Sorry I didn't see the number section.

It did adjust when I clicked on it, can't adjust it back up though

[image]

JacquesdeBruyn, 2023-07-16:
I Just increased the max value and it can now be adjusted back

[image]

syssi, 2023-07-16:
Looks good to me. I will implement a (over/under)temperature setting later the day. So we can finish this project as feature complete.

JacquesdeBruyn, 2023-07-16:
It's interesting that the slider changed to a text box when the range increased. 
I'll keep an eye out for your updates

syssi, 2023-07-16:
I've pushed a big bunch of additional `number` entities. Could you verify/test one of the temperature settings?

JacquesdeBruyn, 2023-07-16:
I'll check in a bit. We can do the same with the seplos bms if you want. I have the pc software for that one as well. Haven't spent too much time trying to find it for this bms

syssi, 2023-07-17:
I've pushed new `text_sensor`s to decode different flags: `warnings, protections, errors`

JacquesdeBruyn, 2023-07-17:
Haven't even gotten to test the last updates

JacquesdeBruyn, 2023-07-17:
> I've pushed new `text_sensor`s to decode different flags: `warnings, protections, errors`

Can't test these but everything else seems functional. I changed a couple random settings back and forth and they seemed to respond properly

[17 more replies]

closed 2023-07-17 by syssi