syssi/esphome-pace-bms #8: Verify Gobel Power RN150 support

syssi, 2023-07-16:
@FCLFCL82 Let's discuss your device model here.

FCLFCL82, 2023-07-16:
> @FCLFCL82 Let's discuss your device model here.

ok getting responses again, log file below.  Also, these are the register addresses (trial and error and assumptions) where i got the readings the first time:

0x0015 SOC
0x0016 Pack Voltage
0x0020 SOH
0x0021 Charge Voltage
0x0025 Max cell voltage
0x0026 Min cell voltge
0x0027 Max volt cell
0x0028 Min volt cell
0x0029 No cells in pack

[13:16:51][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms current': Sending state 0.00000 A with 2 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms power': Sending state 0.00000 W with 2 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 16.00
[13:16:51][D][sensor:094]: 'pace-bms total voltage': Sending state 0.16000 V with 2 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 16.00
[13:16:51][D][sensor:094]: 'pace-bms state of charge': Sending state 16.00000 % with 0 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms state of health': Sending state 0.00000 % with 0 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 16.00
[13:16:51][D][sensor:094]: 'pace-bms remaining capacity': Sending state 0.16000 Ah with 2 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms Full capacity': Sending state 0.00000 Ah with 2 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 16.00
[13:16:51][D][sensor:094]: 'pace-bms Design capacity': Sending state 0.16000 Ah with 2 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[13:16:51][D][uart_debug:114]: <<< 01:03:10:00:00:00:10:00:10:00:00:00:10:00:00:00:10:00:00:9E:0D
[13:16:51][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms warnings bitmask': Sending state 0.00000  with 0 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms protection bitmask': Sending state 0.00000  with 0 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[13:16:51][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:51][D][sensor:094]: 'pace-bms balancer status': Sending state 0.00000  with 0 decimals of accuracy
[13:16:52][D][uart_debug:114]: <<< 01:03:08:00:00:00:00:00:00:00:00:95:D7
[13:16:52][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:52][D][sensor:094]: 'pace-bms cell voltage 1': Sending state 0.00000 V with 3 decimals of accuracy
[13:16:52][D][:465]: Cell voltage 1: 0.000 V
[13:16:52][D][:465]: Cell voltage 2: 0.000 V
[13:16:52][D][:465]: Cell voltage 3: 0.000 V
[13:16:52][D][:465]: Cell voltage 4: 0.000 V
[13:16:52][D][:465]: Cell voltage 5: 1.313 V
[13:16:52][D][:465]: Cell voltage 6: 0.000 V
[13:16:52][D][:465]: Cell voltage 7: 0.099 V
[13:16:52][D][:465]: Cell voltage 8: 5.301 V
[13:16:52][D][:465]: Cell voltage 9: 0.000 V
[13:16:52][D][:465]: Cell voltage 10: 0.017 V
[13:16:52][D][:465]: Cell voltage 11: 0.000 V
[13:16:52][D][:465]: Cell voltage 12: 19.850 V
[13:16:52][D][:465]: Cell voltage 13: 20.000 V
[13:16:52][D][:465]: Cell voltage 14: 0.257 V
[13:16:52][D][:465]: Cell voltage 15: 0.036 V
[13:16:52][D][:465]: Cell voltage 16: 0.000 V
[13:16:52][D][sensor:094]: 'pace-bms average cell voltage': Sending state 2.92956 V with 3 decimals of accuracy
[13:16:52][D][sensor:094]: 'pace-bms min cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[13:16:52][D][sensor:094]: 'pace-bms max cell voltage': Sending state 20.00000 V with 3 decimals of accuracy
[13:16:52][D][sensor:094]: 'pace-bms min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[13:16:52][D][sensor:094]: 'pace-bms max voltage cell': Sending state 13.00000  with 0 decimals of accuracy
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 20.00
[13:16:52][D][sensor:094]: 'pace-bms delta cell voltage': Sending state 20.00000 V with 3 decimals of accuracy
[... 24 lines cut ...]
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 257.00
[13:16:52][D][sensor:094]: 'pace-bms cell voltage 14': Sending state 0.25700 V with 3 decimals of accuracy
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 36.00
[13:16:52][D][sensor:094]: 'pace-bms cell voltage 15': Sending state 0.03600 V with 3 decimals of accuracy
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:52][D][sensor:094]: 'pace-bms cell voltage 16': Sending state 0.00000 V with 3 decimals of accuracy
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[13:16:52][D][sensor:094]: 'pace-bms battery temperature 1': Sending state 0.00000 ˚C with 1 decimals of accuracy
[13:16:52][D][modbus_controller.sensor:025]: Sensor new state: 100.00
[... cut ...]
[13:16:52][D][uart_debug:114]: <<< 01:03:2C:00:00:00:00:00:00:00:00:05:21:00:00:00:63:14:B5:00:00:00:11:00:00:4D:8A:4E:20:01:01:00:24:00:00:00:00:00:64:15:90:00:00:27:10:00:00:3B:B9
[13:16:52][D][uart_debug:114]: >>> 01:03:00:69:00:02:14:17
[13:16:52][D][modbus.number:023]: Number new state : 0.00
[13:16:52][D][number:012]: 'pace-bms balance start cell voltage': Sending state 0.000000
[13:16:52][D][modbus.number:023]: Number new state : 0.00
[13:16:52][D][number:012]: 'pace-bms balance start delta voltage': Sending state 0.000000
[13:16:52][D][uart_debug:114]: <<< 01:03:04:00:00:00:00:FA:33

syssi, 2023-07-16:
I wasn't aware you are using different register addresses. Interesting! Did you use the `PbmsTools.exe` in the past? I would be happy about a screenshot of the `System Config` tab. Escially the `Inverter protocol` section. I assume the selected protocol isn't called `PACE_MODBUS` in your case.

syssi, 2023-07-16:
You could use this example configuration to dump register 0 to 200. May be we can identify some more registers.

https://github.com/syssi/esphome-pace-bms/blob/main/tests/esp8266-modbus-dump.yaml

FCLFCL82, 2023-07-16:
> You could use this example configuration to dump register 0 to 200. May be we can identify some more registers.
> 
> https://github.com/syssi/esphome-pace-bms/blob/main/tests/esp8266-modbus-dump.yaml

I get the following errors when i run that:

`[22:17:38][E][modbus:160]: send too many values 200 max=128
[22:17:38][E][modbus:160]: send too many values 200 max=128
[22:17:38][E][modbus:160]: send too many values 200 max=128
[22:17:38][E][modbus:160]: send too many values 200 max=128
[22:17:39][E][modbus:160]: send too many values 200 max=128
[22:17:39][D][modbus_controller:032]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue`

FCLFCL82, 2023-07-16:
> I wasn't aware you are using different register addresses. Interesting! Did you use the `PbmsTools.exe` in the past? I would be happy about a screenshot of the `System Config` tab. Escially the `Inverter protocol` section. I assume the selected protocol isn't called `PACE_MODBUS` in your case.

I use a too called RfnBMS_Tools to interface with the BMS, it looks identical to pbmstools but that doesnt work.  This is from the system parameters page.  

[image]

syssi, 2023-07-17:
Could you set the RS485 protocol to `ModbusC1636` and verify the already identified registers are still working? I assume the default protocol (`None`) is equal to the first item (`ModbusC1636`).

syssi, 2023-07-17:
> [22:17:38][E][modbus:160]: send too many values 200 max=128

I've updated the YAML to fix this issue. Please give it another try with the updated version:

https://github.com/syssi/esphome-pace-bms/blob/main/tests/esp8266-modbus-dump.yaml

FCLFCL82, 2023-07-17:
I will have to change the modbus later when home.  I am getting the same errors with the dump yaml.  i reduced the second section to 20 registers and got the following results:

```[11:10:13][D][uart_debug:114]: >>> 01:03:00:00:00:78:45:E8
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! 9089!=63
[11:10:13][D][uart_debug:114]: <<< 01:03:F0:00:00:00:10:00:10:00:00:00:10:00:00:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:10:00:20:00:20:00:21:00:21:00:21:00:20:00:21:00:21:00:20:00:20:00:20:00:20:00:21:00:21:00:20:00:21:00:00:00:00:00:00:00:00:00:06:0B:5C:0B:5C:0B:5C:0B:5B:0B:57:0B:72:00:00:00:00:00:00:00:00:00:00:00:00:02:11:00:63:00:03:0D:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:03:0D:40:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! F523!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][W][modbus:108]: Modbus CRC Check failed! C071!=00
[11:10:13][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:48:01:C0:00:00:05:DC:00:C0:00:00:00:00:00:00:31:9C
[11:10:13][D][uart_debug:114]: >>> 01:03:00:00:00:78:45:E8
[11:10:13][D][uart_debug:114]: <<< 01:03:F0:00:00:00:10:00:10:00:00:00:10:00:00:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:10:00:20:00:20:00:21:00:21:00:21:00:20:00:21:00:21:00:20:00:20:00:20:00:20:00:21:00:21:00:20:00:21:00:00:00:00:00:00:00:00:00:06:0B:5C:0B:5C:0B:5C:0B:5C:0B:57:0B:72:00:00:00:00:00:00:00:00:00:00:00:00:02:11:00:63:00:03:0D:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:03:0D:40:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[11:10:13][D][text_sensor:064]: 'pace-bms dump address 0-99': Sending state '000000100010000000100000001000000000000000000000000000000000000000100020002000210021002100200021002100200020002000200021002100200021000000000000000000060b5c0b5c0b5c0b5c0b570b720000000000000000000000000211006300030d400000000000000000000000000000000000030d40001000000000000000000000000000000000000000000000000000000000000000060000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[11:10:13][D][text_sensor:064]: 'pace-bms dump address 100-199': Sending state '0211006300030d400000000000000000000000000000000000030d40001000000000000000000000000000000000000000000000000000000000000000060000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000024801c0000005dc00c0000000000000'
[11:10:13][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:48:01:C0:00:00:05:DC:00:C0:00:00:00:00:00:00:95:9F```

syssi, 2023-07-17:
```
  addr  value
     0  0x0000
     1  0x0010
     2  0x0010
     3  0x0000
     4  0x0010
     5  0x0000
     6  0x0010
     7  0x0000
     8  0x0000
     9  0x0000
    10  0x0000
    11  0x0000
    12  0x0000
    13  0x0000
    14  0x0000
    15  0x0000
    16  0x0010
    17  0x0020
    18  0x0020
    19  0x0021
    20  0x0021
    21  0x0021
    22  0x0020
    23  0x0021
    24  0x0021
    25  0x0020
    26  0x0020
    27  0x0020
    28  0x0020
    29  0x0021
    30  0x0021
    31  0x0020
    32  0x0021
    33  0x0000
    34  0x0000
    35  0x0000
    36  0x0000
    37  0x0006
    38  0x0B5C
    39  0x0B5C
    40  0x0B5C
    41  0x0B5B
    42  0x0B57
    43  0x0B72
    44  0x0000
    45  0x0000
    46  0x0000
    47  0x0000
    48  0x0000
    49  0x0000
    50  0x0211
    51  0x0063
    52  0x0003
    53  0x0D40
    54  0x0000
    55  0x0000
    56  0x0000
    57  0x0000
    58  0x0000
    59  0x0000
    60  0x0000
    61  0x0000
    62  0x0003
    63  0x0D40
    64  0x0010
    65  0x0000
    66  0x0000
    67  0x0000
    68  0x0000
    69  0x0000
    70  0x0000
    71  0x0000
    72  0x0000
    73  0x0000
    74  0x0000
    75  0x0000
    76  0x0000
    77  0x0000
    78  0x0000
    79  0x0000
    80  0x0006
    81  0x0000
    82  0x0000
    83  0x0000
    84  0x0000
    85  0x0000
    86  0x0000
    87  0x0000
    88  0x0000
    89  0x0000
    90  0x0000
    91  0x0000
    92  0x0000
    93  0x0000
    94  0x0000
    95  0x0000
    96  0x0000
    97  0x0000
    98  0x0000
    99  0x0000
   100  0x0000
   101  0x0000
   102  0x0000
   103  0x0000
   104  0x0000
   105  0x0000
   106  0x0000
   107  0x0000
   108  0x0000
   109  0x0000
   110  0x0000
   111  0x0000
   112  0x0248
   113  0x01C0
   114  0x0000
   115  0x05DC
   116  0x00C0
   117  0x0000
   118  0x0000
   119  0x0000
```

The response doesn't look plausible.

syssi, 2023-07-17:
Let's request the registers one by one as next try. I will prepare another YAML.

syssi, 2023-07-17:
I've prepared another YAML which prints the response to the log: https://github.com/syssi/esphome-pace-bms/blob/rn150-poke-around/tests/esp8266-modbus-dump.yaml

Please give it a try. If the response doesn't look good again I will prepare the "request one register per request" attempt.

FCLFCL82, 2023-07-17:
Here's the output from that one:

```[15:24:34][D][uart_debug:114]: >>> 01:03:00:00:00:77:05:EC
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=1102
[15:24:34][W][modbus:108]: Modbus CRC Check failed! 3059!=D03
[15:24:34][W][modbus:108]: Modbus CRC Check failed! 1470!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=D03
[15:24:34][D][uart_debug:114]: <<< 01:03:EE:00:00:00:10:00:10:00:00:00:10:00:00:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:10:00:20:00:20:00:21:00:21:00:21:00:20:00:21:00:21:00:20:00:20:00:20:00:20:00:21:00:21:00:20:00:21:00:00:00:00:00:00:00:00:00:06:0B:60:0B:5E:0B:5E:0B:5E:0B:5F:0B:79:00:00:00:00:00:00:00:00:00:00:00:00:02:11:00:63:00:03:0D:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:03:0D:40:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! 2E3C!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=06
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][W][modbus:108]: Modbus CRC Check failed! C071!=00
[15:24:34][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:48:01:C0:00:00:05:DC:00:C0:00:00:00:00:5A:FD
[15:24:34][D][uart_debug:114]: >>> 01:03:00:00:00:77:05:EC
[15:24:34][D][uart_debug:114]: <<< 01:03:EE:00:00:00:10:00:10:00:00:00:10:00:00:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:10:00:20:00:20:00:21:00:21:00:21:00:20:00:21:00:21:00:20:00:20:00:20:00:20:00:21:00:21:00:20:00:21:00:00:00:00:00:00:00:00:00:06:0B:60:0B:5E:0B:5D:0B:5E:0B:5F:0B:79:00:00:00:00:00:00:00:00:00:00:00:00:02:11:00:63:00:03:0D:40:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:03:0D:40:00:10:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[15:24:34][I][:068]: Register 0x00 (0): 0x0000 (0)
[15:24:34][I][:068]: Register 0x01 (1): 0x0010 (16)
[15:24:34][I][:068]: Register 0x02 (2): 0x0010 (16)
[15:24:34][I][:068]: Register 0x03 (3): 0x0000 (0)
[15:24:34][I][:068]: Register 0x04 (4): 0x0010 (16)
[15:24:34][I][:068]: Register 0x05 (5): 0x0000 (0)
[15:24:34][I][:068]: Register 0x06 (6): 0x0010 (16)
[15:24:34][I][:068]: Register 0x07 (7): 0x0000 (0)
[15:24:34][I][:068]: Register 0x08 (8): 0x0000 (0)
[15:24:34][I][:068]: Register 0x09 (9): 0x0000 (0)
[15:24:34][I][:068]: Register 0x0A (10): 0x0000 (0)
[15:24:34][I][:068]: Register 0x0B (11): 0x0000 (0)
[15:24:34][I][:068]: Register 0x0C (12): 0x0000 (0)
[15:24:34][I][:068]: Register 0x0D (13): 0x0000 (0)
[15:24:34][I][:068]: Register 0x0E (14): 0x0000 (0)
[15:24:34][I][:068]: Register 0x0F (15): 0x0000 (0)
[15:24:34][I][:068]: Register 0x10 (16): 0x0010 (16)
[15:24:34][I][:068]: Register 0x11 (17): 0x0020 (32)
[15:24:34][I][:068]: Register 0x12 (18): 0x0020 (32)
[15:24:34][I][:068]: Register 0x13 (19): 0x0021 (33)
[15:24:34][I][:068]: Register 0x14 (20): 0x0021 (33)
[15:24:34][I][:068]: Register 0x15 (21): 0x0021 (33)
[15:24:34][I][:068]: Register 0x16 (22): 0x0020 (32)
[15:24:34][I][:068]: Register 0x17 (23): 0x0021 (33)
[15:24:34][I][:068]: Register 0x18 (24): 0x0021 (33)
[15:24:34][I][:068]: Register 0x19 (25): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1A (26): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1B (27): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1C (28): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1D (29): 0x0021 (33)
[15:24:34][I][:068]: Register 0x1E (30): 0x0021 (33)
[15:24:34][I][:068]: Register 0x1F (31): 0x0020 (32)
[15:24:34][I][:068]: Register 0x20 (32): 0x0021 (33)
[15:24:34][I][:068]: Register 0x21 (33): 0x0000 (0)
[15:24:34][I][:068]: Register 0x22 (34): 0x0000 (0)
[15:24:34][I][:068]: Register 0x23 (35): 0x0000 (0)
[15:24:34][I][:068]: Register 0x24 (36): 0x0000 (0)
[15:24:34][I][:068]: Register 0x25 (37): 0x0006 (6)
[15:24:34][I][:068]: Register 0x26 (38): 0x0B60 (2912)
[15:24:34][I][:068]: Register 0x27 (39): 0x0B5E (2910)
[15:24:34][I][:068]: Register 0x28 (40): 0x0B5D (2909)
[... 56 lines cut ...]
[15:24:35][I][:068]: Register 0x61 (97): 0x0000 (0)
[15:24:35][I][:068]: Register 0x62 (98): 0x0000 (0)
[15:24:35][I][:068]: Register 0x63 (99): 0x0000 (0)
[15:24:35][I][:068]: Register 0x64 (100): 0x0000 (0)
[15:24:35][I][:068]: Register 0x65 (101): 0x0000 (0)
[15:24:35][I][:068]: Register 0x66 (102): 0x0000 (0)
[15:24:35][I][:068]: Register 0x67 (103): 0x0000 (0)
[15:24:35][I][:068]: Register 0x68 (104): 0x0000 (0)
[15:24:35][I][:068]: Register 0x69 (105): 0x0000 (0)
[15:24:35][I][:068]: Register 0x6A (106): 0x0000 (0)
[... cut ...]
[15:24:35][I][:068]: Register 0x76 (118): 0x0000 (0)
[15:24:35][D][modbus_controller.sensor:025]: Sensor new state: 1.00
[15:24:35][D][sensor:093]: 'pace-bms read up to 119': Sending state 1.00000  with 0 decimals of accuracy
[15:24:35][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:48:01:C0:00:00:05:DC:00:C0:00:00:00:00:CE:FC```

syssi, 2023-07-17:
These could be temperatures:

```
[15:24:34][I][:068]: Register 0x26 (38): 0x0B60 (2912)
[15:24:34][I][:068]: Register 0x27 (39): 0x0B5E (2910)
[15:24:34][I][:068]: Register 0x28 (40): 0x0B5D (2909)
[15:24:35][I][:068]: Register 0x29 (41): 0x0B5E (2910)
[15:24:35][I][:068]: Register 0x2A (42): 0x0B5F (2911)
[15:24:35][I][:068]: Register 0x2B (43): 0x0B79 (2937)
```

Is your environment temperature around 29°C. Does this fit?

syssi, 2023-07-17:
Some corresponding screenshots of the `RfnBMS_Tools` (all tabs) would be awesome!

FCLFCL82, 2023-07-17:
temps are approx 20C.  i will get some screen shots later, i can also send you a link for the software if it helps on this and other projects.  I think my next step should be try all of the RS485 protocols on the working YAML, hopefully one of them works.  Is the Gobel Power repository of any use? i couldn't find any references to register address

JacquesdeBruyn, 2023-07-17:
This chunk maybe cell voltages?

[15:24:34][I][:068]: Register 0x11 (17): 0x0020 (32)
[15:24:34][I][:068]: Register 0x12 (18): 0x0020 (32)
[15:24:34][I][:068]: Register 0x13 (19): 0x0021 (33)
[15:24:34][I][:068]: Register 0x14 (20): 0x0021 (33)
[15:24:34][I][:068]: Register 0x15 (21): 0x0021 (33)
[15:24:34][I][:068]: Register 0x16 (22): 0x0020 (32)
[15:24:34][I][:068]: Register 0x17 (23): 0x0021 (33)
[15:24:34][I][:068]: Register 0x18 (24): 0x0021 (33)
[15:24:34][I][:068]: Register 0x19 (25): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1A (26): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1B (27): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1C (28): 0x0020 (32)
[15:24:34][I][:068]: Register 0x1D (29): 0x0021 (33)
[15:24:34][I][:068]: Register 0x1E (30): 0x0021 (33)
[15:24:34][I][:068]: Register 0x1F (31): 0x0020 (32)
[15:24:34][I][:068]: Register 0x20 (32): 0x0021 (33)

Values should be bigger but these bigger sized batteries typically have either 15 or 16 cells

FCLFCL82, 2023-07-17:
We are making progress, protocol changed to ModbusC1636, log dump below and some interpretations:

```[23:13:25][I][:068]: Register 0x00 (0): 0x0000 (0)
[23:13:25][I][:068]: Register 0x01 (1): 0x14AA (5290) Pack voltage
[23:13:25][I][:068]: Register 0x02 (2): 0x0063 (99) SOC
[23:13:25][I][:068]: Register 0x03 (3): 0x0064 (100) SOH
[23:13:25][I][:068]: Register 0x04 (4): 0x4D67 (19815) Remaining capacity (198150mAH)
[23:13:25][I][:068]: Register 0x05 (5): 0x4E20 (20000) Full capacity (200000mAH)
[23:13:25][I][:068]: Register 0x06 (6): 0x4E20 (20000) Design capacity (200000mAH)
[23:13:25][I][:068]: Register 0x07 (7): 0x0000 (0) --possibly charge cycles
[23:13:25][I][:068]: Register 0x08 (8): 0x0000 (0)
[23:13:25][I][:068]: Register 0x09 (9): 0x0000 (0)
[23:13:25][I][:068]: Register 0x0A (10): 0x0000 (0)
[23:13:25][I][:068]: Register 0x0B (11): 0x0800 (2048) ??
[23:13:25][I][:068]: Register 0x0C (12): 0x0000 (0)
[23:13:25][I][:068]: Register 0x0D (13): 0x0000 (0)
[23:13:25][I][:068]: Register 0x0E (14): 0x0000 (0)
[23:13:25][I][:068]: Register 0x0F (15): 0x0CDF (3295) C1 voltage
[23:13:25][I][:068]: Register 0x10 (16): 0x0CDB (3291) c2
[23:13:25][I][:068]: Register 0x11 (17): 0x0CE8 (3304) c3
[23:13:25][I][:068]: Register 0x12 (18): 0x0CF2 (3314) c4
[23:13:25][I][:068]: Register 0x13 (19): 0x0CFA (3322) c5
[23:13:25][I][:068]: Register 0x14 (20): 0x0CDD (3293) c6
[23:13:25][I][:068]: Register 0x15 (21): 0x0CE6 (3302) c7
[23:13:25][I][:068]: Register 0x16 (22): 0x0CEF (3311) c8
[23:13:25][I][:068]: Register 0x17 (23): 0x0CE0 (3296) c9
[23:13:25][I][:068]: Register 0x18 (24): 0x0CDE (3294) c10
[23:13:25][I][:068]: Register 0x19 (25): 0x0CDC (3292) c11
[23:13:25][I][:068]: Register 0x1A (26): 0x0CE0 (3296) c12
[23:13:25][I][:068]: Register 0x1B (27): 0x0CF9 (3321) c13
[23:13:25][I][:068]: Register 0x1C (28): 0x0CF0 (3312) c14
[23:13:25][I][:068]: Register 0x1D (29): 0x0CD7 (3287) c15
[23:13:25][I][:068]: Register 0x1E (30): 0x0CE7 (3303) c16
[23:13:25][I][:068]: Register 0x1F (31): 0x00BD (189) cell 1 temp 18.9
[23:13:25][I][:068]: Register 0x20 (32): 0x00BB (187) cell 2 temp
[23:13:25][I][:068]: Register 0x21 (33): 0x00BB (187) cell 3 temp
[23:13:25][I][:068]: Register 0x22 (34): 0x00BD (189) cell 4 temp
[23:13:25][I][:068]: Register 0x23 (35): 0x00C2 (194) mosfet temp
[23:13:25][I][:068]: Register 0x24 (36): 0x00DE (222) ambient temp
[23:13:25][I][:068]: Register 0x25 (37): 0x0248 (584) charge cut off voltage 58.4V
[23:13:25][I][:068]: Register 0x26 (38): 0x0000 (0) 
[23:13:25][I][:068]: Register 0x27 (39): 0x05DC (1500) ??
[23:13:25][I][:068]: Register 0x28 (40): 0x0000 (0)
[23:13:25][I][:068]: Register 0x29 (41): 0x0000 (0)
[23:13:25][I][:068]: Register 0x2A (42): 0x0000 (0)
[23:13:25][I][:068]: Register 0x2B (43): 0x0000 (0)
[23:13:25][I][:068]: Register 0x2C (44): 0x0000 (0)
[23:13:25][I][:068]: Register 0x2D (45): 0x0000 (0)
[23:13:25][I][:068]: Register 0x2E (46): 0x0000 (0)
[23:13:25][I][:068]: Register 0x2F (47): 0x0000 (0)
[23:13:25][I][:068]: Register 0x30 (48): 0x0000 (0)
[23:13:25][I][:068]: Register 0x31 (49): 0x0000 (0)
[23:13:25][I][:068]: Register 0x32 (50): 0x0000 (0)
[23:13:25][I][:068]: Register 0x33 (51): 0x0000 (0)
[23:13:25][I][:068]: Register 0x34 (52): 0x0000 (0)
[23:13:25][I][:068]: Register 0x35 (53): 0x0000 (0)
[23:13:25][I][:068]: Register 0x36 (54): 0x0000 (0)
[23:13:25][I][:068]: Register 0x37 (55): 0x0000 (0)
[23:13:25][I][:068]: Register 0x38 (56): 0x0000 (0)
[23:13:25][I][:068]: Register 0x39 (57): 0x0000 (0)
[23:13:25][I][:068]: Register 0x3A (58): 0x0000 (0)
[23:13:25][I][:068]: Register 0x3B (59): 0x0000 (0)
[23:13:25][I][:068]: Register 0x3C (60): 0xDDE0 (56800) Pack OVP Alarm
[23:13:25][I][:068]: Register 0x3D (61): 0xE420 (58400) Pack OVP
[23:13:25][I][:068]: Register 0x3E (62): 0xD7A0 (55200) Pack OVP recovery
[23:13:25][I][:068]: Register 0x3F (63): 0x001E (30) Pack OVP Delay (3000ms)
[23:13:25][I][:068]: Register 0x40 (64): 0x0E10 (3600) Cell OVP Alarm
[23:13:25][I][:068]: Register 0x41 (65): 0x0E42 (3650) Cell OVP
[23:13:25][I][:068]: Register 0x42 (66): 0x0D7A (3450) Cell OVP recovery
[23:13:25][I][:068]: Register 0x43 (67): 0x000A (10) Delay (1000ms)
[... 26 lines cut ...]
[23:13:25][I][:068]: Register 0x5E (94): 0xFF38 (65336)
[23:13:25][I][:068]: Register 0x5F (95): 0xFF9C (65436)
[23:13:25][I][:068]: Register 0x60 (96): 0x0384 (900)
[23:13:25][I][:068]: Register 0x61 (97): 0x047E (1150)
[23:13:25][I][:068]: Register 0x62 (98): 0x0352 (850)
[23:13:25][I][:068]: Register 0x63 (99): 0x01C2 (450)
[23:13:25][I][:068]: Register 0x64 (100): 0x0384 (900)
[23:13:25][I][:068]: Register 0x65 (101): 0x0352 (850)
[23:13:25][I][:068]: Register 0x66 (102): 0xFFA6 (65446)
[23:13:25][I][:068]: Register 0x67 (103): 0xFF06 (65286)
[23:13:25][I][:068]: Register 0x68 (104): 0xFF9C (65436)
[23:13:25][I][:068]: Register 0x69 (105): 0x0D48 (3400)
[23:13:25][I][:068]: Register 0x6A (106): 0x001E (30)
[23:13:25][I][:068]: Register 0x6B (107): 0xE420 (58400)
[23:13:25][I][:068]: Register 0x6C (108): 0x1388 (5000)
[23:13:25][I][:068]: Register 0x6D (109): 0x0C4E (3150)
[23:13:25][I][:068]: Register 0x6E (110): 0x05A0 (1440)
[23:13:25][I][:068]: Register 0x6F (111): 0x0004 (4)
[23:13:25][I][:068]: Register 0x70 (112): 0x000A (10)
[23:13:25][I][:068]: Register 0x71 (113): 0x0000 (0)
[23:13:25][I][:068]: Register 0x72 (114): 0x0000 (0)
[23:13:25][I][:068]: Register 0x73 (115): 0x0000 (0)
[23:13:25][I][:068]: Register 0x74 (116): 0x0000 (0)
[23:13:25][I][:068]: Register 0x75 (117): 0x0000 (0)
[23:13:25][I][:068]: Register 0x76 (118): 0x0000 (0)```

JacquesdeBruyn, 2023-07-17:
Looks good, just that high voltage cut off.... Feeling very aggressive 😅 hope you're not planning to charge that high

FCLFCL82, 2023-07-17:
> Looks good, just that high voltage cut off.... Feeling very aggressive 😅 hope you're not planning to charge that high

:) my inverter limits to 100A thankfully!  getting readouts now also:

```[23:42:57][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[23:42:57][D][sensor:094]: 'gobelp-bms current': Sending state 0.00000 A with 2 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[23:42:57][D][sensor:094]: 'gobelp-bms power': Sending state 0.00000 W with 2 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 5290.00
[23:42:57][D][sensor:094]: 'gobelp-bms total voltage': Sending state 52.90000 V with 2 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 99.00
[23:42:57][D][sensor:094]: 'gobelp-bms state of charge': Sending state 99.00000 % with 0 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 100.00
[23:42:57][D][sensor:094]: 'gobelp-bms state of health': Sending state 100.00000 % with 0 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 19815.00
[23:42:57][D][sensor:094]: 'gobelp-bms remaining capacity': Sending state 198.14999 Ah with 2 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 20000.00
[23:42:57][D][sensor:094]: 'gobelp-bms full capacity': Sending state 200.00000 Ah with 2 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 20000.00
[23:42:57][D][sensor:094]: 'gobelp-bms design capacity': Sending state 200.00000 Ah with 2 decimals of accuracy
[23:42:57][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[23:42:57][D][sensor:094]: 'gobelp-bms charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[23:42:57][D][uart_debug:114]: <<< 01:03:10:00:00:14:AA:00:63:00:64:4D:67:4E:20:4E:20:00:00:48:C5
[23:42:58][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[23:42:58][D][sensor:094]: 'gobelp-bms warnings bitmask': Sending state 0.00000  with 0 decimals of accuracy
[23:42:58][D][text_sensor:064]: 'gobelp-bms warnings': Sending state ''
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[23:42:58][D][sensor:094]: 'gobelp-bms protections bitmask': Sending state 0.00000  with 0 decimals of accuracy
[23:42:58][D][text_sensor:064]: 'gobelp-bms protections': Sending state ''
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 2048.00
[23:42:58][D][sensor:094]: 'gobelp-bms errors bitmask': Sending state 2048.00000  with 0 decimals of accuracy
[23:42:58][D][text_sensor:064]: 'gobelp-bms errors': Sending state ''
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[23:42:58][D][sensor:094]: 'gobelp-bms balancer status': Sending state 0.00000  with 0 decimals of accuracy
[23:42:58][D][uart_debug:114]: <<< 01:03:08:00:00:00:00:08:00:00:00:97:B7
[23:42:58][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3295.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 1': Sending state 3.29500 V with 3 decimals of accuracy
[23:42:58][D][sensor:094]: 'gobelp-bms average cell voltage': Sending state 3.30200 V with 3 decimals of accuracy
[23:42:58][D][sensor:094]: 'gobelp-bms min cell voltage': Sending state 3.28700 V with 3 decimals of accuracy
[23:42:58][D][sensor:094]: 'gobelp-bms max cell voltage': Sending state 3.32200 V with 3 decimals of accuracy
[23:42:58][D][sensor:094]: 'gobelp-bms min voltage cell': Sending state 15.00000  with 0 decimals of accuracy
[23:42:58][D][sensor:094]: 'gobelp-bms max voltage cell': Sending state 5.00000  with 0 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 0.04
[23:42:58][D][sensor:094]: 'gobelp-bms delta cell voltage': Sending state 0.03500 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3291.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 2': Sending state 3.29100 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3304.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 3': Sending state 3.30400 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3314.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 4': Sending state 3.31400 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3322.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 5': Sending state 3.32200 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3293.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 6': Sending state 3.29300 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3302.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 7': Sending state 3.30200 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3311.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 8': Sending state 3.31100 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3296.00
[23:42:58][D][sensor:094]: 'gobelp-bms cell voltage 9': Sending state 3.29600 V with 3 decimals of accuracy
[23:42:58][D][modbus_controller.sensor:025]: Sensor new state: 3294.00
[... cut ...]
[23:42:59][D][modbus.number:023]: Number new state : 0.00
[23:42:59][D][number:012]: 'gobelp-bms Charging overcurrent 2 protection delay time': Sending state 0.000000
[23:42:59][D][uart_debug:114]: <<< 01:03:6E:DD:E0:E4:20:D7:A0:00:1E:0E:10:0E:42:0D:7A:00:0A:B5:40:AF:00:C4:E0:00:1E:0B:22:0A:BE:0C:1C:00:0A:00:9B:00:A0:00:0A:00:9B:00:A0:00:1E:01:2C:00:06:01:F4:02:26:01:F4:02:26:02:58:02:26:00:32:00:00:00:32:FF:9C:FF:38:FF:9C:03:84:04:7E:03:52:01:C2:03:84:03:52:FF:A6:FF:06:FF:9C:0D:48:00:1E:E4:20:13:88:0C:4E:05:A0:00:04:00:0A:00:00:00:00:FA:D0```

JacquesdeBruyn, 2023-07-17:
I was referring to voltage(58.4) something around 56-56.4 is a more sensible area unless you really need ro rapid charge. I've noticed that they get to 100% soc as low as 54.5v. That pack is 200Ah right? So should be able to handle 200A charge without any problems.

FCLFCL82, 2023-07-17:
Ah yes, I was changing a few parameters to check the logs. Normally set it to 55.2v

syssi, 2023-07-18:
Good job! Please contribute your YAML at the end. Did you identify the `current` register?

FCLFCL82, 2023-07-18:
i used the pace YAML you provided, i havent made any changes and it seems to work.  I am getting this in the middle of the logs if it means anything:

```[08:43:38][D][modbus_controller.sensor:025]: Sensor new state: 202.00
[08:43:38][D][sensor:094]: 'gobelp-bms environment temperature': Sending state 20.20000 ˚C with 1 decimals of accuracy
[08:43:38][D][uart_debug:114]: <<< 01:03:2C:0C:DF:0C:DA:0C:E5:0C:EE:0C:FA:0C:DC:0C:E3:0C:EA:0C:DE:0C:DC:0C:DB:0C:DF:0C:F4:0C:EC:0C:D6:0C:E6:00:B6:00:B4:00:B6:00:B6:00:AE:00:CA:13:76
[08:43:38][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[08:43:38][D][modbus.number:023]: Number new state : 56800.00
[08:43:38][D][number:012]: 'gobelp-bms pack overvoltage alarm': Sending state 56800.000000

JacquesdeBruyn, 2023-07-18:
Over voltage alarm? What voltage is your pack at?

FCLFCL82, 2023-07-18:
the two uart_debug:114 lines, is that normal?

JacquesdeBruyn, 2023-07-18:
Yes, it shows the command being sent and the response received

FCLFCL82, 2023-07-18:
thanks.  what determines the update interval, i know there is `update_interval: 10s` in the YAML but some of the sensors update every hour, some minutes and some every 10s.

JacquesdeBruyn, 2023-07-18:
Yes, it shows the command being sent and the response received it updates everything every 10s. But it will only registers updates that have changed. So if your soc for instance stays at 98 for an hour it would show the last change as an hour ago

FCLFCL82, 2023-07-18:
that makes sense, thanks for your help on this. @syssi thank you for your work, the support you provide is better than the BMS suppliers!  keep it up.  i now have your seplos-bms, neey-balancer (3no. by bluetooth) and pace/gobelp-bms projects running.

closed 2023-07-18 by syssi