syssi/esphome-pace-bms #16: Configuration does not work for Delong batteries

Algram, 2023-09-22:
Hi,

I have a battery pack from "Titan Solar" which uses the Pace BMS and Delong batteries.
I have no idea on how to make this communicate with the ESP32. I am using an RS485 converter with flow control, but I never get a response back. It seems like the RS485 is disabled or something?

There are also different protocols selectable compared to other pace bms versions out there. Do you know which would be the right one for Modbus?

[image]

Algram, 2023-09-24:
I got it working for the first pack by switching to a different version of the PBMS Tools software. This one had the protocol "Pace-Modbus" selectable.

How can I access multiple battery modules? I tried different slave addresses, but they are not found: 
[image]

hhoebel, 2024-02-06:
As fas as i understand the RS485-connection of the Pace-BMS, you only have access to the BMS to which you are connected. That means, you have to connect all your batteries with RS485 to read them out. You can either use one ESP32 per battery or connect all batteries togehter (use a simple RJ45-splitter with 1:1 connection at every battery and connect with the next battery) and add the additional batteries in the code of one ESP. In this case, it should work as listed above (address 0x1, address 0x2 etc.)

ThHanika, 2024-04-06:
The RS485 protocol has to be PACE. If it isn't selectable, you have to take an other PC-software
[image]

My  software release: PbmsTools HS.1.0.7

syssi, 2024-05-10:
This issue could be resolved by using an different version of the PBMS Tools software. The `PACE_MODBUS` protocol was available here.

closed 2024-05-10 by syssi