syssi/esphome-pace-bms #17: PACE BMS P16S200A

MattBN001, 2023-10-02:
Hi,

can anybody help with the PACE BMS P16S200A?

log-output is attached.
Protocol is pace_modbus

[image]

syssi, 2023-10-02:
Could you provide the traffic as plaintext? The screenshot isn't easily readable.

MattBN001, 2023-10-02:
Sorry, sure:

[12:34:21][D][modbus_controller:043]: Modbus command to device=1 register=0x3C countdown=0 no response received - removed from send queue
[12:34:21][D][uart_debug:114]: <<< 7E:32:35:30:46:34:36:39:30:30:30:30:30:46:44:39:30:0D
[12:34:21][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:39:30:30:30:30:30:46:44:41:34:0D
[12:34:21][D][uart_debug:114]: <<< 7E:32:35:30:33:34:36:39:30:30:30:30:30:46:44:41:33:0D
[12:34:21][D][uart_debug:114]: <<< 7E:32:35:30:34:34:36:39:30:30:30:30:30:46:44:41:32:0D
[12:34:21][D][uart_debug:114]: <<< 7E:32:35:30:35:34:36:39:30:30:30:30:30:46:44:41:31:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:39:34:36:39:30:30:30:30:30:46:44:39:44:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[12:34:22][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D
[12:34:23][D][uart_debug:114]: <<< 7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[12:34:23][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[12:34:23][D][uart_debug:114]: <<< 7E:32:35:30:46:34:36:39:30:30:30:30:30:46:44:39:30:0D
[12:34:23][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:39:30:30:30:30:30:46:44:41:34:0D
[12:34:23][D][uart_debug:114]: <<< 7E:32:35:30:33:34:36:39:30:30:30:30:30:46:44:41:33:0D
[12:34:23][D][uart_debug:114]: <<< 7E:32:35:30:34:34:36:39:30:30:30:30:30:46:44:41:32:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:35:34:36:39:30:30:30:30:30:46:44:41:31:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:39:34:36:39:30:30:30:30:30:46:44:39:44:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
[12:34:24][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:46:34:36:39:30:30:30:30:30:46:44:39:30:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:32:34:36:39:30:30:30:30:30:46:44:41:34:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:33:34:36:39:30:30:30:30:30:46:44:41:33:0D
[12:34:25][D][uart_debug:114]: <<< 7E:32:35:30:34:34:36:39:30:30:30:30:30:46:44:41:32:0D
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:35:34:36:39:30:30:30:30:30:46:44:41:31:0D
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[12:34:26][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:39:34:36:39:30:30:30:30:30:46:44:39:44:0D
[12:34:26][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:41:34:36:39:30:30:30:30:30:46:44:39:35:0D
[12:34:26][D][uart_debug:114]: <<< 7E:32:35:30:42:34:36:39:30:30:30:30:30:46:44:39:34:0D
[12:34:27][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:27][D][uart_debug:114]: <<< 7E:32:35:30:43:34:36:39:30:30:30:30:30:46:44:39:33:0D
[12:34:27][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:27][D][uart_debug:114]: <<< 7E:32:35:30:44:34:36:39:30:30:30:30:30:46:44:39:32:0D
[12:34:27][D][uart_debug:114]: <<< 7E:32:35:30:45:34:36:39:30:30:30:30:30:46:44:39:31:0D
[12:34:27][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:27][D][modbus_controller:043]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[12:34:27][D][uart_debug:114]: <<< 7E:32:35:30:46:34:36:39:30:30:30:30:30:46:44:39:30:0D
[12:34:27][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[12:34:27][W][modbus:108]: Modbus CRC Check failed! A901!=8282
[12:34:27][W][modbus:108]: Modbus CRC Check failed! 2931!=511
[12:34:27][D][uart_debug:114]: <<< FB:FF:9F:82:82:82:82:82:11:05:0D
[12:34:27][D][uart_debug:114]: <<< 7E:32:35:30:33:34:36:39:30:30:30:30:30:46:44:41:33:0D
[12:34:27][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[12:34:28][D][uart_debug:114]: <<< 7E:32:35:30:34:34:36:39:30:30:30:30:30:46:44:41:32:0D
[12:34:28][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[12:34:28][D][uart_debug:114]: <<< 7E:32:35:30:35:34:36:39:30:30:30:30:30:46:44:41:31:0D
[12:34:28][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[12:34:28][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[12:34:28][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[12:34:28][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[12:34:28][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[... cut ...]
[12:34:36][D][uart_debug:114]: <<< 7E:32:35:30:36:34:36:39:30:30:30:30:30:46:44:41:30:0D
[12:34:36][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:36][D][uart_debug:114]: <<< 7E:32:35:30:37:34:36:39:30:30:30:30:30:46:44:39:46:0D
[12:34:37][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[12:34:37][D][uart_debug:114]: <<< 7E:32:35:30:38:34:36:39:30:30:30:30:30:46:44:39:45:0D
[12:34:37][D][uart_debug:114]: <<< 7E:32:35:30:39:34:36:39:30:30:30:30:30:46:44:39:44:0D
[12:34:37][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C

syssi, 2023-11-30:
@MattBN001 The selected protocol is wrong. Your device does respond with Seplos frames at the moment: 

```
~250E46900000FD91
~250F46900000FD90
~250246900000FDA4
~250346900000FDA3
~250446900000FDA2
~250546900000FDA1
~250646900000FDA0
~250746900000FD9F
~250846900000FD9E
~250946900000FD9D
~250A46900000FD95
~250B46900000FD94
~250C46900000FD93
```

Please use the tool `pbmstools` to select another Modbus based protocol. I'm talking about this setting:

[image]

syssi, 2023-11-30:
> Protocol is pace_modbus

If it's already the case could you provide a photo of the different communication ports of the battery pack? Do you use the RS485 port for sure?