syssi/esphome-pace-bms #19: Help for P16S100A, looks like PACE

Maikeleg, 2023-11-28:
Hello,
I have read the topic/issue "Add PACE BMS support #1" and tried with the default "esp32-example.yaml" but can't get communication to work. My MQTT is working as i can see in the MQTT explorer. I got my battery in error with all experiments (luckily solved now) so I'm a bit more carefull now.
When i readout the pack with pbmstools i can only make connection with the version "Gobel_PC-BMS_Tools_HS2.0.3_Win32", i use rs485 on the parralel port. dip switch is set to 0, the dedicated rs485 next to can isnt working, other dipswitch settings are not working
However I see that address 0x01 is setup in the yaml. Do I need to change the pack dipswitch or change the yaml?
Also I use and only have the "esp32doit-devkit-v1" which i setup in yaml as: 
```
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
```
is this correct? I see in the topic "Add PCA BMS support" only 8266 mentioned and a different board.
Also in your documentation you say to use orange/orange-white for B/A, however with pbmstools i can only communicate with brown/brown-white as B/A
I'm using a Joyvoit Suns Energy battery pack BW5KW: https://www.alibaba.com/product-detail/Energy-Power-wall-5-3-2kw_1600330104517.html
Using the pbmstools i get the VER readout: P16S100A-21468-1.00
Searching on google I think this is a PACE BMS, it also look like one, I can confirm it has 100A and 16 cells
Also I'm using the following RS485 3.3-5V module: https://a.aliexpress.com/_mt820MY

Is changing the address setting enough to make this work, and if so do I have to change the dipswitch on the pack or the setting in the yaml?
Sorry for all questions, I'm not experienced with ESP but would really like this to work
Greetings Maikel

syssi, 2023-11-28:
1. Could you provide a photo of your RS485 converter Module?
2. Which GPIOs do you use as UART pins? Does the TX LED of the RS485 board blink periodically?
3. You could use the ESP+RS485 converter to capture some traffic between the BMS and the workstation. I would like to make sure the correct protocol (Modbus frames) are used here.
4. Yes. Have have to align the device address on both sides. If the BMS uses `0x00` you have to use the same address in the YAML.
5. You can ignore the `esp32.board` setting if you assign GPIOs at the YAML using `GPIOxx`. If you want to use the labels on the PCB like `D1, D2, D2,...` the setting is important.
6. Is there a drop-down menu available in pbmstools to choose between different protocols?

Maikeleg, 2023-11-28:
Thanks a lot for your reply, i'll gather all information and come back to you as soon as possible

Maikeleg, 2023-11-28:
> 1. Could you provide a photo of your RS485 converter Module?
> 
>     2. Which GPIOs do you use as UART pins? Does the TX LED of the RS485 board blink periodically?
> 
>     3. You could use the ESP+RS485 converter to capture some traffic between the BMS and the workstation. I would like to make sure the correct protocol (Modbus frames) is used here.
> 
>     4. Yes. Have have to align the device address on both sides. If the BMS uses `0x00` you have to use the same address in the YAML.
> 
>     5. You can ignore the `esp32.board` setting if you assign GPIOs at the YAML using `GPIOxx`. If you want to use the labels on the PCB like `D1, D2, D2,...` the setting is important.
> 
>     6. Is there a drop-down menu available in pbmstools to choose between different protocols?
1) This board;
[image]
[image]
[image]
2) GPIO 16+17 which should be UART 2 RX and UART 2 TX which i connected to RX-RX and TX-TX to the converter board
3) Could you tell me how to do that, connect the esp32 to usb-PC with converter board connected? 
4) I will try to change the yaml address and upload to the esp32 again later today 
5) Aah ok thanks
6) no there is not, it chooses that itself as it seems. Right under ADDR (Address) is shows
Protocol: None, HS-PC-232-BP-V1.1 (it is connected through RS485)

I'll check address and give you an update later today

ThHanika, 2023-11-28:
The Modbus seems to need a configurated address to work, I use AddressNo. 1. (switch 1 set ON)
This example does the job. ( My impediments was wrong PIN-Selections on ESP-Board ;-P )

Maikeleg, 2023-11-28:
> The Modbus seems to need a configurated address to work, I use AddressNo. 1. (switch 1 set ON) This example does the job. ( My impediments was wrong PIN-Selections on ESP-Board ;-P )
Ok thanks for your reply, I will set both to address 1. No time this evening but will try tomorrow morning

ThHanika, 2023-11-30:
I use pbmstools with interface RS232. RS485 is the monitoring interface for me. With homeassistant and esphome, i can also change the parameter (but not all)

Maikeleg, 2023-11-30:
> This sounds like good progress. You have to change the address at the YAML to `0x01` too. Otherwise the responses doesn't get picked up and decoded.

I also dont see any value in all the mqtt topics so I'm afraid its not being decoded
I also look at the topic in grafana via the mqtt plugin and see all information but again no value

Maikeleg, 2023-11-30:
> FYI! The modbus-address 0x00 is a broadcast address and is not a valid slave-address

Do you mean by Slave address the battery pack address?? I only have one battery pack

Maikeleg, 2023-11-30:
> The responses aren't valid Modbus frames. It looks like Seplos/Pylon responses. This reason could be one of the following options
> 
>     1. You've selected the wrong protocol at the RS485 protocol drop-down
> 
>     2. You are using the wrong port
> 
>     3. The BMS firmware doesn't support Modbus communication

Thanks for your quick reply
1. I cannot change the protocol settings (in pbmstools i think you mean?)
2. I tried all rs485 ports
3. I will ask the battery manufacturer if that's the case

Do you know by any chance if the seplos/pylon method also needs to be on pack 1? I tried that but only with dip 0
Changing both pack dip and yaml setting to 0x00 not going to work anyway or worth a try?

I'll check issue 17 now! Thanks a lot

Maikeleg, 2023-11-30:
> This issue (#17) is about to the same problem.

Ok thanks again, I will try that tomorrow morning and let you know

Maikeleg, 2023-12-02:
> I use pbmstools with interface RS232. RS485 is the monitoring interface for me. With homeassistant and esphome, i can also change the parameter (but not all)

I'm a bit in doubt what to do next. With whatever version of pbmstools I tried yesterday I cannot change the inverter setting. Most versions only work with pack 1-xx. The one that works for me is the Gobel 2.03HS. Can write parameters, readout current stats but cannot change inverter protocol. Seems that's only possible with a rs232??
Thing is that connector isn't on any of my pcs so have to buy another adapter for that. No problem but will cost quite some time
I have 8 different pbmstools to give you an idea and I'm running out of ideas to be honest
Thanks for your and the developpers help so far!! Much appreciated

syssi, 2023-12-02:
Could you provide a screenshot of the `pbmstools` protocols section? Did you try to contact the seller / manufacturer?

ThHanika, 2023-12-02:
... and you now the **special secret** password (123456) for changing settings?

Maikeleg, 2023-12-02:
> Could you provide a screenshot of the `pbmstools` protocols section? Did you try to contact the seller / manufacturer?

I'll make tomorrow
Planning to contact the factory in Monday. Let you know tmr

Maikeleg, 2023-12-02:
> ... and you now the **special secret** password (123456) for changing settings?

Yups have that. Is a specific password for other settings and one for inverter protocol

ThHanika, 2023-12-02:
Look at andy's video canal:
https://www.youtube.com/watch?v=TbUUCXPN1Xg

Maikeleg, 2023-12-05:
> Could you provide a screenshot of the `pbmstools` protocols section? Did you try to contact the seller / manufacturer?

[image]
[image]
[image]
[image]
[image]
[image]
If i change the xml file i also have an option with modbus protocol, i changed it back recently as one og the things to try
So as you can see i can readout all info except read or write inverter protocol
In the video that @ThHanika  Thomas posted (thanks a lot for that) i understand i need an rs232 cable

I sent a message to the company i bought the battery from, didn't got an answer yet, will ask them again

ThHanika, 2023-12-05:
On the last picture, you have to click read under **Inverter protocol** to show selectet protocols

Maikeleg, 2023-12-05:
> On the last picture, you have to click <Read> under **Inverter protocol** to show selected protocols
> 
> BTW: First you have generaly click <Read> before you can write any parameters, Pace expect complete filled parameterfield to update its parameter.

Hi Thomas,
I just talked to the lady in china who supports the battery, she says i need another firmware version and (ofcourse) discouraged me to change the setting. I'm waiting for her to email me. Keep you updated

Maikeleg, 2023-12-07:
Just ordered these two cables, one with chipset according to a diysolar topic and one for SOK battery
[image]

Maikeleg, 2024-01-12:
I got all cable the day before going on holiday and all three of them didnt work out of the box
Since then I was out of country and busy getting back with things, I'll plan to continue this project probably monday morning and will report back
Supposedly i need a special driver for the direct connected cables so i have some things to try still

Maikeleg, 2024-02-25:
I got it working with the following cable: SOK SK48V100 Battery RS232 serial communicatrion Prolific USB RS232 to RJ11 6P6C
I needed to update the serial driver for the prolific chipset PL2303GT on windows
Then set protocol to PACE_MODBUS in special version of Pbmstools that i got from factory 
After that i kept using this cable to readout the battery
Thanks for all help @syssi and @ThHanika

closed 2024-05-10 by syssi

theyo-tester, 2024-06-24:
@Maikeleg

May I Know which PBMSTool Version you used? Is it online somewhere available?

Thanks!

Maikeleg, 2024-09-22:
> @Maikeleg
> 
> May I know which PBMSTool Version you used? Is it online somewhere available? My versions does not allow me to change the protocols....
> 
> Thanks!

So sorry I just read your reply now;
I got his version "PbmsTools HS1.0.6£¨¿ÉÑ¡Ð­Òé£©" which I'm almost 100% sure is the same as the version HS1.06
If you want it please let me know, i can sent it to you.
What makes the difference in being able to change protocol settings is using the RS232 interface with the right cable.
See above which i ordered ("Just ordered these two cables, one with chipset according to a diysolar topic and one for SOK battery") One of these is working (they look exactly the same so don't know which one)

I ended up keeping using the rs232 cable because it works reliably and i had a usb-port left :)

tuxcrafter, 2025-03-24:
Could you please share a link to the Gobel_PC-BMS_Tools_HS2.0.3_Win32.zip tool? and did you get ESPHOME working, I got an P16S100A-21247-2.00 that I would like to communicate with.