syssi/esphome-pace-bms #20: Wrong Bitmask for status charging and discharging

ThHanika, 2023-11-28:
please correct your code example [esp32-example.yaml](https://github.com/syssi/esphome-pace-bms/blob/main/esp32-example.yaml) :

```
binary_sensor:
  #  11  Status/Fault flag                     2 byte   R  uint16  Hex See ^3
  #      Bit 8: Status: Charging enabled/disabled
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} charging"
    address: 11
    register_type: holding
    bitmask: 0x0100

  #      Bit 9: Status: Discharging enabled/disabled
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} discharging"
    address: 11
    register_type: holding
    bitmask: 0x0200
```

BTW: This status is a working status. For enabled charging/discharging it is a better choice to use MOSFET-status 

Your Example is tested and worked on **Pace BMS PC16S200A Firmwareversion 3.0**

syssi, 2023-11-28:
Thanks for your feedback! Do you like to provide a pull request to contribute your improvement?

ThHanika, 2023-11-28:
OK,  I clone your project

closed 2024-05-10 by syssi