syssi/esphome-pace-bms #23: Help for Titansolar with P16S100A-21037-3.00T

yutani42, 2023-12-03:
Hi there,
I got a small 2.5 kWh battery which id' like to read the soc from.
So I tried your setup, but sadly I don't get a response from the thing.
Some screenshots:
[image]
[image]
[image]
[image]
Logs:
`[C][uart.arduino_esp8266:102]: UART Bus:
[C][uart.arduino_esp8266:103]:   TX Pin: GPIO5
[C][uart.arduino_esp8266:104]:   RX Pin: GPIO4
[C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[C][uart.arduino_esp8266:109]:   Data Bits: 8
[C][uart.arduino_esp8266:110]:   Parity: NONE
[C][uart.arduino_esp8266:111]:   Stop bits: 1
[C][uart.arduino_esp8266:115]:   Using software serial
[C][modbus:143]: Modbus:
[C][modbus:145]:   Send Wait Time: 200 ms
[C][modbus:146]:   CRC Disabled: NO
[D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[C][modbus_controller:298]: ModbusController:
[C][modbus_controller:299]:   Address: 0x01
[D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[D][modbus_controller:040]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8
[D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8
[D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8
[D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8
[D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8
[D][modbus_controller:040]: Modbus command to device=1 register=0x0B countdown=0 no response received - removed from send queue`

(shorted the log outputs a bit)

I hope I didn't made a stupid mistake with the wiring, I tried mutliple variants already (e.g. tx/rx on GPIO1/3).

Thanks for any tips.

yutani42, 2023-12-07:
Hi, thanks for the input,
1. I used the following pinout plan for reference:
[image]
Here D1/D2  correspond to GPIO5/4
2. for the rs485 to ttl module:
[image]
Here is DI and RO correspond to TX/RX (see https://microcontrollerslab.com/rs485-serial-communication-esp32-esp8266-tutorial/)

syssi, 2023-12-07:
You have identified the GPIOs of your dev board correctly. `D1 = GPIO5`, `D2 = GPIO4`. @ThHanika was on the wrong track. 

But there is something else you have to care about! The mentioned RS485 converter board requires active flow control. How did you connect the RE/DE pins to your ESP and did you use the `flow_control` setting at your YAML correctly?

yutani42, 2023-12-07:
Well.. I have no clue about a flow control :(
Nothing about that in my yaml... a quick search https://esphome.io/search.html?q=flow+control does not reveal anything..
please help :)

syssi, 2023-12-07:
This schematics shows the different RS485 converter modules. The first one with automatic flow control and the second one (yours!) with manual flow control. In other words: The ESP must press the push-to-talk button on TX (transmission).

https://github.com/syssi/esphome-solax-x1-mini#schematics

Please connect the `DE+RE` pins and attach the pin to a free GPIO of your ESP. To convince ESPHome to care about flow control you have to extend your YAML configuration:

```
modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms
    flow_control_pin: GPIO0
```

yutani42, 2023-12-10:
Hi, I finally had time to craft a Y-cable, and it works... kind of..
I can read the temperatures, but nothing else?

[image]

Logs:
´´´[18:15:26][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[18:15:26][W][modbus:108]: Modbus CRC Check failed! 90AA!=8C14
[18:15:26][D][uart_debug:114]: <<< FF:01:03:10:FA:8D:14:8C:00:48:00:64:0E:A6:14:4D:13:88:00:2C:99:4A:00
[18:15:26][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[18:15:26][W][modbus:108]: Modbus CRC Check failed! 90AA!=8C14
[18:15:27][D][uart_debug:114]: <<< FF:01:03:10:FA:8D:14:8C:00:48:00:64:0E:A6:14:4D:13:88:00:2C:99:4A:00
[18:15:27][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[18:15:27][W][modbus:108]: Modbus CRC Check failed! D32A!=8C14
[18:15:27][D][uart_debug:114]: <<< FF:01:03:10:FA:77:14:8C:00:48:00:64:0E:A6:14:4D:13:88:00:2C:E3:09:00
[18:15:27][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[18:15:27][W][modbus:108]: Modbus CRC Check failed! D32A!=8C14
[18:15:27][D][uart_debug:114]: <<< FF:01:03:10:FA:77:14:8C:00:48:00:64:0E:A6:14:4D:13:88:00:2C:E3:09:00
[18:15:27][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[18:15:27][W][modbus:108]: Modbus CRC Check failed! 52EB!=8C14
[18:15:27][D][uart_debug:114]: <<< FF:01:03:10:FA:8A:14:8C:00:48:00:64:0E:A6:14:4D:13:88:00:2C:DE:48:00
[18:15:27][D][modbus_controller:040]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[18:15:27][D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8
[18:15:27][W][modbus_controller:064]: Modbus device=1 back online
[18:15:27][D][uart_debug:114]: <<< 01:03:02:0E:00:BC:24:00
[18:15:28][D][uart_debug:114]: >>> 01:03:00:23:00:02:35:C1
[18:15:28][W][modbus:108]: Modbus CRC Check failed! D669!=E000
[18:15:28][D][uart_debug:114]: <<< FF:01:03:04:00:F4:00:E0:BA:49:00
[18:15:28][D][uart_debug:114]: >>> 01:03:00:23:00:02:35:C1
[18:15:28][W][modbus:108]: Modbus CRC Check failed! D97D!=E000
[18:15:28][D][uart_debug:114]: <<< 00:01:03:04:00:F4:00:E0:BA:49:00
[18:15:28][D][uart_debug:114]: >>> 01:03:00:23:00:02:35:C1
[18:15:28][D][modbus_controller.sensor:025]: Sensor new state: 244.00
[18:15:28][D][sensor:093]: 'TitanSolar mosfet temperature': Sending state 24.40000 ˚C with 1 decimals of accuracy
[18:15:28][D][modbus_controller.sensor:025]: Sensor new state: 224.00
[18:15:28][D][sensor:093]: 'TitanSolar environment temperature': Sending state 22.40000 ˚C with 1 decimals of accuracy´´´

yutani42, 2023-12-13:
It seems this problem's cause lies deeper within arduino, cf https://github.com/esphome/issues/issues/3912
Thanks for helping with the original problem, I'll close this here.

closed 2023-12-13 by yutani42

Adminius, 2023-12-24:
@yutani42 can you please post your yaml with titansolar register? And what Y cable you are talking about?

yutani42, 2023-12-25:
Hi @Adminius ,
the registers are all from esp8266-example.yaml (I commented some of them out), nothing new here.
For the Y-cable, I am not sure if this is term only exists in germany (where I am from), I meant a forked cable, just as @syssi described in https://github.com/syssi/esphome-solax-x1-mini#schematics (the second one)

@syssi fyi: I resolved my crc issues by removing the common ground. I am not sure whats going on there, but both the battery and the esp are isolated, so a common ground (for cable shielding) seems futile (or even harmful..)

There is just one slight issue left: the modbus communication fails when the battery is charging.. (but only then)