syssi/esphome-pace-bms #25: Can't enable RS485

clydebarrow, 2024-01-11:
I have a Jakiper 48V100 battery with Pace BMS. When I connect to it with pbmstools I can see all the data, but the CANBus and RS485 protocol selection drop-downs are empty. Clicking Read shows a "successful" message, but still does not populate the list, and there is a a red "interruption" message in the status bar lower right.

Any ideas? Is there more than one variation of the pbmstools software?

I have the password, and I have tried th Jakiper tool, but it has no selection for the RS485 protocol.

syssi, 2024-01-11:
Could you try to talk to your manufacturer and ask about another firmware with RS485 support for your battery bank? This is just an idea. I don't know how to make sure RS485 is supported or not.

TreshanAyesh, 2024-01-25:
Hi, we successfully connected and read data from a battery to the serial monitor. But there is an issue with mqtt not connecting. Does it have something to do with home assistant? Can we use a mqtt broker to connect to node red?

syssi, 2024-01-28:
The `mqtt` component of ESPHome is able talk to any MQTT broker and can be used with any Home Automation software / it's not limited to Home Assistant. Could you attach a USB cable to your ESP board and provide a ESPHome log? It will supply all needed details whats going on.

TreshanAyesh, 2024-01-29:
[image]
This is the error we get ehrn connecting to mqtt. the logger level is set to very verbose. The mqtt component is asking for a IP but we are not much sure about this ip

syssi, 2024-01-29:
Could you provide your YAML configuration too? It looks like the configuration of the `mqtt` section is wrong.

TreshanAyesh, 2024-01-29:
this is the yaml file we use to test the mqtt alone now

syssi, 2024-05-10:
Please pass the FQDN or IP address here:

```
mqtt:
  broker: 192.168.1.1
  port: 1883
  username: !secret mqtt_username
  password: !secret mqtt_password
```

MQTT-URLs aren't supported here.

closed 2024-05-10 by syssi