syssi/esphome-pace-bms #27: PACE via

tumbochka, 2024-02-18:


tumbochka, 2024-02-18:
Just no response:
```
[21:40:29][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:29][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:30][D][modbus_controller:032]: Modbus command to device=1 register=0x0F countdown=0 no response received - removed from send queue
[21:40:30][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:30][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:30][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:30][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:31][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:31][D][modbus_controller:032]: Modbus command to device=1 register=0x3C countdown=0 no response received - removed from send queue
[21:40:36][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[21:40:36][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[21:40:36][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[21:40:37][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[21:40:37][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[21:40:37][D][modbus_controller:032]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[21:40:37][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[21:40:37][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[21:40:38][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[21:40:38][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[21:40:38][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[21:40:38][D][modbus_controller:032]: Modbus command to device=1 register=0x09 countdown=0 no response received - removed from send queue
[21:40:38][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:38][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:39][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:39][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:39][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[21:40:39][D][modbus_controller:032]: Modbus command to device=1 register=0x0F countdown=0 no response received - removed from send queue
[21:40:39][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:40][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:40][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:40][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:40][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[21:40:40][D][modbus_controller:032]: Modbus command to device=1 register=0x3C countdown=0 no response received - removed from send queue
```

The config (addresses etc) is the same as for seplos.

tumbochka, 2024-02-18:
My pace config (not working):

```
substitutions:
  name: bms
  friendly_name: Anern Battery BMS
  device_description: "Monitor a Seplos BMS on Anern battery via RS232"
  tx_pin: GPIO23
  rx_pin: GPIO34

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

binary_sensor:
  #  11  Status/Fault flag                     2 byte   R  uint16  Hex See ^3
  #      Bit 8: Status: Charging enabled/disabled
  - platform: modbus_controller
[... 1536 lines cut ...]
    address: 150
    register_type: holding
    register_count: 10
    response_size: 20

  # 160  Model SN                             20 byte  RW  uint16  ASCII BMS Manufacturer
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} bms serial number"
    skip_updates: 60
    address: 160
    register_type: holding
    register_count: 10
    response_size: 20

  # 170  PACK SN                              20 byte  RW  uint16  ASCII PACK Manufacturer
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} battery pack serial number"
    skip_updates: 60
    address: 170
    register_type: holding
    register_count: 10
    response_size: 20 
```

tumbochka, 2024-02-18:
My seplos config (working):

```
substitutions:
  name: bms
  friendly_name: Anern Battery BMS
  device_description: "Monitor a Seplos BMS on Anern battery via RS232"
  external_components_source: github://syssi/esphome-seplos-bms@main
  battery_bank0: "${name} bank 0"
  battery_bank1: "${name} bank 1"
  tx_pin: GPIO23
  rx_pin: GPIO34

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

api:

uart:
  id: uart_0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384
  debug:
    dummy_receiver: false
    direction: BOTH

seplos_modbus:
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms

seplos_bms:
  - id: battery_bank0
    # Dip switch configuration of a single pack setup / address 0x0
    #  8    7    6    5    4    3    2    1
    # off, off, off, off, off, off, off, off
    address: 0x01
    # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
    protocol_version: 0x20
    seplos_modbus_id: modbus0
    update_interval: 10s

sensor:
  - platform: seplos_bms
    seplos_bms_id: battery_bank0
    min_cell_voltage:
      name: "${battery_bank0} min cell voltage"
    max_cell_voltage:
      name: "${battery_bank0} max cell voltage"
    min_voltage_cell:
      name: "${battery_bank0} min voltage cell"
    max_voltage_cell:
      name: "${battery_bank0} max voltage cell"
    delta_cell_voltage:
      name: "${battery_bank0} delta cell voltage"
    average_cell_voltage:
      name: "${battery_bank0} average cell voltage"
    cell_voltage_1:
      name: "${battery_bank0} cell voltage 1"
    cell_voltage_2:
      name: "${battery_bank0} cell voltage 2"
    cell_voltage_3:
      name: "${battery_bank0} cell voltage 3"
    cell_voltage_4:
      name: "${battery_bank0} cell voltage 4"
    cell_voltage_5:
      name: "${battery_bank0} cell voltage 5"
    cell_voltage_6:
      name: "${battery_bank0} cell voltage 6"
    cell_voltage_7:
      name: "${battery_bank0} cell voltage 7"
    cell_voltage_8:
      name: "${battery_bank0} cell voltage 8"
    cell_voltage_9:
      name: "${battery_bank0} cell voltage 9"
    cell_voltage_10:
      name: "${battery_bank0} cell voltage 10"
    cell_voltage_11:
      name: "${battery_bank0} cell voltage 11"
    cell_voltage_12:
      name: "${battery_bank0} cell voltage 12"
    cell_voltage_13:
      name: "${battery_bank0} cell voltage 13"
    cell_voltage_14:
      name: "${battery_bank0} cell voltage 14"
    cell_voltage_15:
      name: "${battery_bank0} cell voltage 15"
    temperature_1:
      name: "${battery_bank0} temperature 1"
    temperature_2:
      name: "${battery_bank0} temperature 2"
    temperature_3:
      name: "${battery_bank0} temperature 3"
    temperature_4:
      name: "${battery_bank0} temperature 4"
    temperature_5:
      name: "${battery_bank0} environment temperature"
    temperature_6:
      name: "${battery_bank0} mosfet temperature"
    total_voltage:
      name: "${battery_bank0} total voltage"
      filters:
        - multiply: 0.1
    current:
      name: "${battery_bank0} current"
    power:
      name: "${battery_bank0} power"
      filters:
        - multiply: 0.1
    charging_power:
      name: "${battery_bank0} charging power"
      filters:
        - multiply: 0.1
    discharging_power:
      name: "${battery_bank0} discharging power"
      filters:
        - multiply: 0.1
    residual_capacity:
      name: "${battery_bank0} residual capacity"
    battery_capacity:
      name: "${battery_bank0} battery capacity"
    rated_capacity:
      name: "${battery_bank0} rated capacity"

```

tumbochka, 2024-02-18:
The seplos logs:

```
[21:50:20][D][uart_debug:114]: >>> 7E:32:30:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:35:0D
[21:50:20][I][seplos_bms:030]: Telemetry frame (66 bytes) received
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 1': Sending state 3.27900 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 2': Sending state 3.27800 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 3': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 4': Sending state 3.27800 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 5': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 6': Sending state 3.27700 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 7': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 8': Sending state 3.27900 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 9': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 10': Sending state 3.27900 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 11': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 12': Sending state 3.27900 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 13': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 14': Sending state 3.27900 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 cell voltage 15': Sending state 3.27900 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 min cell voltage': Sending state 3.27700 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 max cell voltage': Sending state 3.28000 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 max voltage cell': Sending state 3.00000  with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 min voltage cell': Sending state 6.00000  with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 delta cell voltage': Sending state 0.00300 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 average cell voltage': Sending state 3.27913 V with 3 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 temperature 1': Sending state 23.30000 °C with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 temperature 2': Sending state 23.20000 °C with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 temperature 3': Sending state 23.40000 °C with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 temperature 4': Sending state 23.40000 °C with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 environment temperature': Sending state 24.40000 °C with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 mosfet temperature': Sending state 23.90000 °C with 0 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 current': Sending state -2.97000 A with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 total voltage': Sending state 49.18700 V with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 power': Sending state -146.08539 W with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 charging power': Sending state 0.00000 W with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 discharging power': Sending state 146.08539 W with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 residual capacity': Sending state 47.76000 Ah with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 battery capacity': Sending state 100.00000 Ah with 2 decimals of accuracy
[21:50:20][D][sensor:094]: 'bms bank 0 rated capacity': Sending state 100.00000 Ah with 2 decimals of accuracy
[21:50:20][W][component:204]: Component seplos_modbus took a long time for an operation (0.38 s).
[21:50:20][W][component:205]: Components should block for at most 20-30ms.
```

dominicknoetze, 2024-02-20:
RS232 + Multiple battery packs would be great.

m1k3f15h, 2024-03-14:
is it posllible to read one Pace BMS via RS232?

Dears Mike

tumbochka, 2024-03-28:
@m1k3f15h This solution doesn't read even one Pace BMS via RS232 - just no response as I showed above. But I can read one Pace BMS via RS232 using Seplos solution https://github.com/syssi/esphome-seplos-bms

m1k3f15h, 2024-03-28:
> @m1k3f15h This solution doesn't read even one Pace BMS via RS232 - just no response as I showed above. But I can read one Pace BMS via RS232 using Seplos solution https://github.com/syssi/esphome-seplos-bms

Thanks for your Answer.

syssi, 2024-05-10:
The BMS supports different protocols which needs to be selected using a windows application called `PbmsTools`. It looks like the Seplos protocol is enabled at the moment. If you want to use this project you have to select `PACE_MODBUS` as the active `RS485 protocol`. See https://github.com/syssi/esphome-pace-bms?tab=readme-ov-file#esphome-pace-bms

closed 2024-05-10 by syssi

tumbochka, 2024-05-12:
@syssi It is pace modbus in the config but looks like it is set for RS485 but not for RS232 - RS232 still works on something seplos-like. 

[image]