syssi/esphome-pace-bms #32: P16S100A with RS232

WallK, 2024-06-08:
Hello!
I have SOK 100Ah battery that uses P16S100A-1B470-3.01 BMS
I'm trying to talk to it using your project, but get no response from the battery
```
[16:41:08][D][uart_debug:158]: >>> "\x01\x03\x00\x01\x00\x01\xD5\xCA"
[16:41:08][D][uart_debug:158]: >>> "\x01\x03\x00\x01\x00\x01\xD5\xCA"
[16:41:08][D][uart_debug:158]: >>> "\x01\x03\x00\x01\x00\x01\xD5\xCA"
[16:41:09][D][uart_debug:158]: >>> "\x01\x03\x00\x01\x00\x01\xD5\xCA"
[16:41:09][D][uart_debug:158]: >>> "\x01\x03\x00\x01\x00\x01\xD5\xCA"
[16:41:09][W][modbus_controller:030]: Modbus device=1 set offline
[16:41:09][D][modbus_controller:043]: Modbus command to device=1 register=0x01 countdown=0 no response received - removed from send queue
```
I can talk to it using the same RS232 port with pbms software

I want to use rs232 because rs485 is taken by inverter and it's set up as pylontech 485

Is it possible to do what I want without disconnecting inverter?

Thank you!

syssi, 2024-06-08:
Did you attach a RS232 to the ESP and did you change the RS232 protocol setting using pbms to Modbus?

WallK, 2024-06-09:
> Did you attach a RS232 to the ESP and did you change the RS232 protocol setting using pbms to Modbus?

I'm attaching ESP to RS232 using MAX3232 line driver. I've did echo test on the output side of the chip and it's working. Also, the same chip is used to communicate with PC

I can't change the protocol on RS232, but it seems to be set to PACE already, see the top of the screenshot below
[image]

syssi, 2024-06-09:
Could you go to the `export data` view and provide an export? It should contain the RAW traffic. May be it will spread some light about the expected payload / used protocol / wrong device address.

WallK, 2024-06-09:
Export data tab is empty
It says "Pick a table from list to export" but there's no tables to pick

Is there a way to sniff com port without blocking it?

syssi, 2024-06-09:
I don't a know an easy way to sniff traffic on a window host without additional hardware.

Please play around a bit more with the Windows app. You should be able to see the raw traffic at some point.

WallK, 2024-06-09:
Played with it and can't find any way to see serial data passed around

Trying to find if it's possible to use another esp as a mitm on UART between uart-usb adapter and uart-rs232
Do you know any projects like this?

syssi, 2024-06-09:
If you are lucky no MITM setup is required. You could just attach an ESP in parallel to the UART-TTL (3.3V level!) pins of your USB converter and sniff the traffic here (using on or two RX pins). 

You can use this YAML example: https://github.com/syssi/esphome-pace-bms/blob/main/tests/esp8266-uart-sniffer.yaml

If you are not lucky both parties aren't happy about the third device on the bus and no communication is possible anymore. As soon the windows application is still able to talk to the BMS you should see the traffic at the log.

WallK, 2024-06-12:
Here's an output, from start of connection 

```
[20:50:58][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:43:31:30:30:30:30:46:44:39:41:0D
[20:50:58][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:36:30:32:38:35:30:33:31:33:36:35:33:33:31:33:30:33:30:34:31:32:44:33:31:34:32:33:34:33:37:33:30:32:44:33:33:32:45:33:30:33:31:32:30:46:35:38:35:0D
[20:50:58][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:43:32:30:30:30:30:46:44:39:39:0D
[20:50:58][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:42:30:35:30:33:31:34:32:33:34:33:37:33:30:33:33:33:30:33:32:33:32:33:35:33:30:33:31:33:35:33:37:33:39:34:34:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:45:45:30:31:0D
[20:50:58][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:30:0D
[20:50:59][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:44:30:37:43:30:30:30:31:31:30:30:43:45:39:30:43:45:41:30:43:45:41:30:43:45:41:30:43:45:41:30:43:45:38:30:43:45:42:30:43:45:37:30:43:45:42:30:43:45:41:30:43:45:41:30:43:45:41:30:43:45:42:30:43:45:43:30:43:45:41:30:43:45:42:30:36:30:42:41:31:30:42:41:34:30:42:39:46:30:42:39:41:30:42:44:45:30:42:45:34:46:43:46:36:43:45:41:30:32:37:32:33:30:34:32:38:32:44:30:30:32:38:32:37:31:30:36:31:45:30:45:34:0D
[20:50:59][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:34:34:45:30:30:32:30:31:46:44:32:45:0D
[20:50:59][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:30:30:34:43:30:30:30:31:31:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:36:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:45:30:30:30:30:30:30:30:30:30:30:30:30:45:46:33:41:0D
[20:50:59][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:43:31:30:30:30:30:46:44:39:41:0D
[20:50:59][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:36:30:32:38:35:30:33:31:33:36:35:33:33:31:33:30:33:30:34:31:32:44:33:31:34:32:33:34:33:37:33:30:32:44:33:33:32:45:33:30:33:31:32:30:46:35:38:35:0D
[20:50:59][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:43:32:30:30:30:30:46:44:39:39:0D
[20:50:59][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:42:30:35:30:33:31:34:32:33:34:33:37:33:30:33:33:33:30:33:32:33:32:33:35:33:30:33:31:33:35:33:37:33:39:34:34:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:45:45:30:31:0D
[20:51:00][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:30:0D
[20:51:00][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:44:30:37:43:30:30:30:31:31:30:30:43:45:39:30:43:45:41:30:43:45:41:30:43:45:42:30:43:45:41:30:43:45:41:30:43:45:42:30:43:45:39:30:43:45:42:30:43:45:42:30:43:45:42:30:43:45:42:30:43:45:42:30:43:45:43:30:43:45:41:30:43:45:42:30:36:30:42:41:31:30:42:41:34:30:42:39:46:30:42:39:41:30:42:44:45:30:42:45:34:46:43:46:34:43:45:41:38:32:37:32:32:30:34:32:38:32:44:30:30:32:38:32:37:31:30:36:31:45:30:44:30:0D
[20:51:00][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:34:34:45:30:30:32:30:31:46:44:32:45:0D
[20:51:00][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:30:30:34:43:30:30:30:31:31:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:36:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:45:30:30:30:30:30:30:30:30:30:30:30:30:45:46:33:41:0D
[20:51:00][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:43:31:30:30:30:30:46:44:39:41:0D
[20:51:00][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:36:30:32:38:35:30:33:31:33:36:35:33:33:31:33:30:33:30:34:31:32:44:33:31:34:32:33:34:33:37:33:30:32:44:33:33:32:45:33:30:33:31:32:30:46:35:38:35:0D
[20:51:01][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:43:32:30:30:30:30:46:44:39:39:0D
[20:51:01][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:30:30:42:30:35:30:33:31:34:32:33:34:33:37:33:30:33:33:33:30:33:32:33:32:33:35:33:30:33:31:33:35:33:37:33:39:34:34:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:32:30:45:45:30:31:0D
```

Longer one is here https://pastebin.com/LekM2JwJ

I've clicked some things, read parameters and some saved memory
There were wifi debug lines, hopefully I've cleaned them up
I think in and out are reversed here

Thanks!

syssi, 2024-06-12:
Good job! I have an idea what's going on. The communication doesn't look like Modbus. It's the Seplos ASCII protocol.

syssi, 2024-06-12:
```
>>> ~250146C10000FD9A
<<< ~25014600602850313653313030412D31423437302D332E303120F585

>>> ~250146C20000FD99
<<< ~25014600B05031423437303330323235303135373944202020202020202020202020202020202020202020202020EE01

>>> ~25014642E00201FD30
<<< ~25014600D07C0001100CE90CEA0CEA0CEA0CEA0CE80CEB0CE70CEB0CEA0CEA0CEA0CEB0CEC0CEA0CEB060BA10BA40B9F0B9A0BDE0BE4FCF6CEA0272304282D0028271061E0E4

>>> ~25014644E00201FD2E
<<< ~25014600004C000110000000000000000000000000000000000600000000000000000000000E000000000000EF3A

>>> ~250146C10000FD9A
<<< ~25014600602850313653313030412D31423437302D332E303120F585

>>> ~250146C20000FD99
<<< ~25014600B05031423437303330323235303135373944202020202020202020202020202020202020202020202020EE01

>>> ~25014642E00201FD30
<<< ~25014600D07C0001100CE90CEA0CEA0CEB0CEA0CEA0CEB0CE90CEB0CEB0CEB0CEB0CEB0CEC0CEA0CEB060BA10BA40B9F0B9A0BDE0BE4FCF4CEA8272204282D0028271061E0D0

>>> ~25014644E00201FD2E
<<< ~25014600004C000110000000000000000000000000000000000600000000000000000000000E000000000000EF3A

>>> ~250146C10000FD9A
<<< ~25014600602850313653313030412D31423437302D332E303120F585

>>> ~250146C20000FD99
<<< ~25014600B05031423437303330323235303135373944202020202020202020202020202020202020202020202020EE01
```

WallK, 2024-06-12:
It really doesn't (:
I've tried config from #27 but it didn't work first try
Should I try again? Should I use your seplos repo?

syssi, 2024-06-12:
Please try this one another time: https://github.com/syssi/esphome-pace-bms/issues/27#issuecomment-1951427401

The first characters of the payloads above does indicate the protocol version `~25`. Please change the config to:

```
protocol_version: 0x25

# instead of

protocol_version: 0x20
```

WallK, 2024-06-12:
OH! Look at this!
```
[21:29:57][D][seplos_modbus:175]: Send frame: ~25014642E00201FD30
[21:29:57][D][uart_debug:114]: >>> 7E:32:35:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:30:0D
[21:29:57][I][seplos_bms:030]: Telemetry frame (68 bytes) received
[21:29:57][D][sensor:094]: 'SOK current': Sending state -6.83000 A with 2 decimals of accuracy
[21:29:57][D][sensor:094]: 'SOK total voltage': Sending state 52.96500 V with 2 decimals of accuracy
[21:29:57][D][sensor:094]: 'SOK power': Sending state -361.75095 W with 2 decimals of accuracy
[21:29:58][D][sensor:094]: 'SOK charging power': Sending state 0.00000 W with 2 decimals of accuracy
[21:29:58][D][sensor:094]: 'SOK discharging power': Sending state 361.75095 W with 2 decimals of accuracy
[21:29:58][D][sensor:094]: 'SOK residual capacity': Sending state 95.85000 Ah with 2 decimals of accuracy
[21:29:58][D][sensor:094]: 'SOK battery capacity': Sending state 102.85000 Ah with 2 decimals of accuracy
[21:29:58][D][sensor:094]: 'SOK rated capacity': Sending state 100.00000 Ah with 2 decimals of accuracy
[21:29:58][W][component:214]: Component seplos_modbus took a long time for an operation (0.07 s).
[21:29:58][W][component:215]: Components should block for at most 20-30ms.
[21:29:58][D][uart_debug:114]: <<< 7E:32:35:30:31:34:36:30:30:44:30:37:43:30:30:30:31:31:30:30:43:45:45:30:43:45:45:30:43:45:46:30:43:45:46:30:43:45:45:30:43:45:46:30:43:45:46:30:43:45:44:30:43:45:45:30:43:45:46:30:43:45:45:30:43:45:46:30:43:45:45:30:43:46:30:30:43:45:44:30:43:45:44:30:36:30:42:41:31:30:42:41:34:30:42:39:46:30:42:39:41:30:42:44:46:30:42:45:34:46:44:35:35:43:45:45:35:32:35:37:31:30:34:32:38:32:44:30:30:32:38:32:37:31:30:35:44:45:30:39:34:0D
```

Yay! Thank you! It could be beneficial to add info about this battery to both readmes

Thanks again!

closed 2024-06-12 by syssi