syssi/esphome-pace-bms #34: Help installing

jacocilliers, 2024-07-03:
Hi
I am a noob and not sure how to get your code into the esp32. Is there any documentation on how I can do the installation on the esp as well as what the pinout is for the esp to ttl please. I would like to integrate it into Home assistant.

syssi, 2024-07-03:
Please start here: https://esphome.io/guides/getting_started_hassio

Try to create/flash your first basic ESPHome node. As soon this is working we will talk about adding this project/implementation to the node.

jacocilliers, 2024-07-03:
Will do thank you

jacocilliers, 2024-07-03:
I have done that part as per the link above. Not sure what to do next? I can see the esp in home assistant and I can edit it but I'm lost.

syssi, 2024-07-03:
Okay. So the node is up and running as `online` node at the dashboard? Do you use a ESP32 or ESP8266?

jacocilliers, 2024-07-03:
Yes that’s correct. ESP32

syssi, 2024-07-03:
Please `edit` the node, don't remove the existing sections because the `ota` encryption key and/or `api` password is important and shouldn't be touched. Please extend YAML using this configuration example:

https://github.com/syssi/esphome-pace-bms/blob/main/esp32-example.yaml

If you provide your basic YAML (excluding the secrets) I can help you getting both files properly merged.

jacocilliers, 2024-07-03:
Can I just copy the code within the home assistant / esphome / edit and paste it here or should it be uploaded for you to edit? I feel like a real idiot currently but this esp32 is winning me by far!

syssi, 2024-07-03:
Please press the `EDIT` button on the ESPHome node tile of the dashboard. You should see here some lines already. Please don't overwrite these lines because they are already uploaded to your ESP and are required for the next over-the-air update. My config example must be added at the end of the existing entries *but* there a no section duplicates allowed. 

If you have trouble to merge both YAMLs properly because it's your first time please provide the YAML code of your new node.

jacocilliers, 2024-07-03:
Hi. I paste the YAML code below. Can you please help me merge the 2 I don't seem to get it working. I removed the encryption key and secret from wifi.

substitutions:
  name: esp-web-tools-example-0f7d6c
  friendly_name: ESP Web Tools Example 0f7d6c
packages:
  esphome.esp_web_tools_example: github://esphome/example-configs/esp-web-tools/esp32.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 

wifi:
  ssid:
  password:

syssi, 2024-07-03:
I've merged both files. You add the `secrets` to your setup or remove the statements again and add you secrets inline:

```
substitutions:
  name: esp-web-tools-example-0f7d6c
  friendly_name: ESP Web Tools Example 0f7d6c
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO16
  rx_pin: GPIO17

packages:
  esphome.esp_web_tools_example: github://esphome/example-configs/esp-web-tools/esp32.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

api:
  encryption:
    key: 

wifi:
  ssid:
  password:

logger:
  level: DEBUG

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

# ...

sensor:
  #   0  Current                               2 byte   R   int16  10mA (Positive: chargingm Negative: discharging)
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} current"
    address: 0
    register_type: holding
    value_type: S_WORD
    unit_of_measurement: "A"
    device_class: current
    state_class: measurement
    accuracy_decimals: 2
[... 335 lines cut ...]
    id: bms0_max_cell_voltage
    name: "${name} max cell voltage"
    update_interval: never
    unit_of_measurement: V
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 3

  - platform: template
    id: bms0_min_voltage_cell
    name: "${name} min voltage cell"
    update_interval: never
    unit_of_measurement: ""
    state_class: measurement
    accuracy_decimals: 0

  - platform: template
    id: bms0_max_voltage_cell
    name: "${name} max voltage cell"
    update_interval: never
    unit_of_measurement: ""
    state_class: measurement
    accuracy_decimals: 0

```

jacocilliers, 2024-07-03:
You are the best thank you. Im still learning and will go through both to teach myself how you did it. Thank you again.

syssi, 2024-07-03:
You have to use these GPIOs:

```
  tx_pin: GPIO16
  rx_pin: GPIO17
```

Feel free to ask additional questions if it doesn't work out of the box.

jacocilliers, 2024-07-03:
Hi. Thank you once again for the help. If my pace bms does not want to work via rs485 can I change it to rs232? I usually use the rs232 to connect to my pc software so at least I know it is working. The pc software gives me an sending failed when trying to change the protocol.

syssi, 2024-07-03:
Which protocols are available via RS232?

jacocilliers, 2024-07-03:
Connecting to Pbms tools gives me options for CAN & rs485 however I am not able to send it and Pace confirmed that the BMS I have cannot be changed ( protocols ) If I connect to solar assistant via rs232 it will connect and show in home assistant however then I cannot see my other 2 battery banks with JK BMS in them, Wanted to see all 3x my batteries in home assistant.

syssi, 2024-07-03:
I don't know which protocols is used on the RS232 port in your case. Just give it a try but don't forget to replace the RS485 converter module with a RS232 (MAX3232) one.

jacocilliers, 2024-07-03:
Will do thank you

syssi, 2024-07-03:
Please take a look at this issue. May be it applies to your setup too: https://github.com/syssi/esphome-pace-bms/issues/32

closed 2024-07-08 by syssi