syssi/esphome-pace-bms #37: It's not bug just a qusetion, how can I identify my bms?

sarpba, 2024-07-07:
I have 3pc chisage ess ce48100-r rack mounted 5KWh battery. I need to disassembly one or other way?
 https://chisagess.com/wp-content/uploads/2023/03/UM_Lithium-Battery-CE48100-R_20230216_V1.2.pdf

syssi, 2024-07-07:
You could just attach an ESP to the RS485 port and give it a try. Do you know any details about the BMS?

syssi, 2024-07-07:
Does the BMS has Bluetooth support? What's the advertised name? Could you ask the seller of your BMS/battery bank about a management software? What's the name of the software and how it looks like?

sarpba, 2024-07-07:
Unfortunatly not have bluetoot connection. I connected it with esp32. With esp32-seplos-v3-example-multiple-battery-banks.yaml config file 
I saw some kind of communication, only in bank2 total voltage report 167.22 V (of course this is false information). 
I'll ask the local reseller for more information tomorrow.

syssi, 2024-07-07:
Feel free to provide some logs including RAW traffic.

sarpba, 2024-07-07:
After a fresh esp install in log:

sarpba, 2024-07-07:
A picture from the batterys
[image]

syssi, 2024-07-07:
Please reduce the YAML configuration to:

```
substitutions:
  name: pace-bms
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: true # <-- this change is super important
```

and provide another log. I will explain what I'm looking for afterwords.

sarpba, 2024-07-07:
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/esphome-web-51a298.yaml...
INFO Starting log output from 192.168.7.109 using esphome API
INFO Successfully connected to esphome-web-51a298 @ 192.168.7.109 in 7.316s
INFO Successful handshake with esphome-web-51a298 @ 192.168.7.109 in 0.071s
[20:31:05][I][app:100]: ESPHome version 2024.6.6 compiled on Jul  7 2024, 20:23:18
[20:31:05][I][app:102]: Project syssi.esphome-pace-bms version 1.0.0
[20:31:05][C][wifi:599]: WiFi:
[20:31:05][C][wifi:427]:   Local MAC: EC:DA:3B:51:A2:98
[20:31:05][C][wifi:432]:   SSID: [redacted]
[20:31:05][C][wifi:435]:   IP Address: 192.168.7.109
[20:31:05][C][wifi:439]:   BSSID: [redacted]
[20:31:05][C][wifi:440]:   Hostname: 'esphome-web-51a298'
[20:31:05][C][wifi:442]:   Signal strength: -43 dB ▂▄▆█
[20:31:05][C][wifi:446]:   Channel: 5
[20:31:05][C][wifi:447]:   Subnet: 255.255.255.0
[20:31:05][C][wifi:448]:   Gateway: 192.168.7.1
[20:31:05][C][wifi:449]:   DNS1: 192.168.7.1
[20:31:05][C][wifi:450]:   DNS2: 0.0.0.0
[20:31:05][C][logger:185]: Logger:
[20:31:05][C][logger:186]:   Level: DEBUG
[20:31:05][C][logger:188]:   Log Baud Rate: 115200
[20:31:05][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[20:31:05][C][uart.idf:139]: UART Bus 1:
[20:31:05][C][uart.idf:140]:   TX Pin: GPIO16
[20:31:05][C][uart.idf:141]:   RX Pin: GPIO17
[20:31:05][C][uart.idf:143]:   RX Buffer Size: 256
[20:31:05][C][uart.idf:145]:   Baud Rate: 9600 baud
[20:31:05][C][uart.idf:146]:   Data Bits: 8
[20:31:05][C][uart.idf:147]:   Parity: NONE
[20:31:05][C][uart.idf:148]:   Stop bits: 1
[20:31:05][C][mdns:115]: mDNS:
[20:31:05][C][mdns:116]:   Hostname: esphome-web-51a298
[20:31:05][C][esphome.ota:073]: Over-The-Air updates:
[20:31:05][C][esphome.ota:074]:   Address: esphome-web-51a298.local:3232
[20:31:05][C][esphome.ota:075]:   Version: 2
[20:31:05][C][safe_mode:018]: Safe Mode:
[20:31:05][C][safe_mode:020]:   Boot considered successful after 60 seconds
[20:31:05][C][safe_mode:021]:   Invoke after 10 boot attempts
[20:31:05][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[20:31:05][C][api:139]: API Server:
[20:31:05][C][api:140]:   Address: esphome-web-51a298.local:6053
[20:31:05][C][api:142]:   Using noise encryption: YES
[20:31:05][D][uart_debug:114]: <<< 0A:45:00:00:00:54:00:00:95:60
[20:31:06][D][uart_debug:114]: <<< 0B:45:00:00:00:54:00:00:54:AC
[20:31:06][D][uart_debug:114]: <<< 0C:45:00:00:00:54:00:00:15:4A
[20:31:07][D][uart_debug:114]: <<< 0D:45:00:00:00:54:00:00:D4:86
[20:31:07][D][uart_debug:114]: <<< 0E:45:00:00:00:54:00:00:94:93
[20:31:08][D][uart_debug:114]: <<< 0F:45:00:00:00:54:00:00:55:5F
[20:31:08][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:00:94:C6
[20:31:08][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:55:00:10:0C:D1:0C:D0:0C:D0:0C:D0:0C:D0:0C:D1:0C:D0:0C:D0:0C:CF:0C:D0:0C:D0:0C:D0:0C:D0:0C:CF:0C:D0:0C:CD:0C:D1:0C:CD:14:7F:FF:FF:FE:2B:0B:E1:0B:E1:27:10:0E:8A:00:44:00:25:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:93:B6
[20:31:09][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:00:55:0A
[20:31:09][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:55:00:10:0C:CD:0C:D0:0C:CE:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CE:0C:D0:0C:CE:0C:D1:0C:CE:0C:D0:0C:D1:0C:CD:14:7E:FF:FF:FE:2D:0B:CD:0B:CD:27:10:0F:58:00:42:00:27:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:F8:07
[20:31:09][D][uart_debug:114]: <<< 04:45:00:00:00:54:00:00:14:EC
[20:31:10][D][uart_debug:114]: <<< 05:45:00:00:00:54:00:00:D5:20
[20:31:10][D][uart_debug:114]: <<< 06:45:00:00:00:54:00:00:95:35
[20:31:11][D][uart_debug:114]: <<< 07:45:00:00:00:54:00:00:54:F9
[20:31:11][D][uart_debug:114]: <<< 08:45:00:00:00:54:00:00:14:B9
[20:31:11][D][uart_debug:114]: <<< 09:45:00:00:00:54:00:00:D5:75
[20:31:12][D][api:102]: Accepted 192.168.7.34
[20:31:12][D][api.connection:1375]: Home Assistant 2024.7.1 (192.168.7.34): Connected successfully
[20:31:12][D][uart_debug:114]: <<< 0A:45:00:00:00:54:00:00:95:60
[20:31:12][D][uart_debug:114]: <<< 0B:45:00:00:00:54:00:00:54:AC
[20:31:13][D][uart_debug:114]: <<< 0C:45:00:00:00:54:00:00:15:4A
[20:31:13][D][uart_debug:114]: <<< 0D:45:00:00:00:54:00:00:D4:86
[20:31:14][D][uart_debug:114]: <<< 0E:45:00:00:00:54:00:00:94:93
[20:31:14][D][uart_debug:114]: <<< 0F:45:00:00:00:54:00:00:55:5F
[20:31:15][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:00:94:C6
[20:31:15][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:55:00:10:0C:D1:0C:CD:0C:D2:0C:CE:0C:D1:0C:CE:0C:D1:0C:CE:0C:D1:0C:CF:0C:D1:0C:CE:0C:D2:0C:CE:0C:D1:0C:CC:0C:D2:0C:CC:14:7F:FF:FF:FE:29:0B:E1:0B:E1:27:10:0E:89:00:44:00:25:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:24:CB
[20:31:15][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:00:55:0A
[... 48 lines cut ...]
[20:31:35][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:55:00:10:0C:CC:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D1:0C:CD:0C:D0:0C:D1:0C:CC:14:7D:FF:FF:FE:2F:0B:CD:0B:CD:27:10:0F:55:00:42:00:27:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:9C:05
[20:31:35][D][uart_debug:114]: <<< 04:45:00:00:00:54:00:00:14:EC
[20:31:35][D][uart_debug:114]: <<< 05:45:00:00:00:54:00:00:D5:20
[20:31:36][D][uart_debug:114]: <<< 06:45:00:00:00:54:00:00:95:35
[20:31:36][D][uart_debug:114]: <<< 07:45:00:00:00:54:00:00:54:F9
[20:31:37][D][uart_debug:114]: <<< 08:45:00:00:00:54:00:00:14:B9
[... cut ...]
[20:31:42][D][uart_debug:114]: <<< 04:45:00:00:00:54:00:00:14:EC
[20:31:42][D][uart_debug:114]: <<< 05:45:00:00:00:54:00:00:D5:20
[20:31:42][D][uart_debug:114]: <<< 06:45:00:00:00:54:00:00:95:35
[20:31:43][D][uart_debug:114]: <<< 07:45:00:00:00:54:00:00:54:F9
[20:31:43][D][uart_debug:114]: <<< 08:45:00:00:00:54:00:00:14:B9
[20:31:44][D][uart_debug:114]: <<< 09:45:00:00:00:54:00:00:D5:75
[20:31:44][D][uart_debug:114]: <<< 0A:45:00:00:00:54:00:00:95:60
[20:31:45][D][uart_debug:114]: <<< 0B:45:00:00:00:54:00:00:54:AC

syssi, 2024-07-07:
Perfect! One BMS acts as Modbus-Master on the Bus and tries to collect the data of 10 other devices on the bus. Two slaves are responding.

syssi, 2024-07-08:
Could you plug the cable into the RS485 port next to the CAN bus als next step? Do you see traffic here too or is there silence as long we don't request data actively from the BMS?

sarpba, 2024-07-08:
I tried, with this truncated config there is silence on that port. In the original "esp32-example.yaml" config file, both LEDs (rx, tx) blink on the modbus converter module.

sarpba, 2024-07-08:
I git this "battery program" from chisage. I'll try it now only.
[Chisage  - Battery program.zip](https://github.com/user-attachments/files/16124699/Chisage.-.Battery.program.zip)

syssi, 2024-07-08:
Could you provide a screenshot of every view? This will help to get an idea whats available.

syssi, 2024-07-08:
Could you ask the seller about a list of the Modbus registers / protocol description?

sarpba, 2024-07-08:
Need a bit time for protocoll (I hope they are give it to me), but there is some log from the program:
[RealTimeData_20240708110602.csv](https://github.com/user-attachments/files/16125307/RealTimeData_20240708110602.csv)
1199 line from battery 1
1200-1241 lines from battery 2
others from battery 3

I can make screensots, if realy need, but the most inportant things in CVS files.

syssi, 2024-07-08:
I would be happy about screenshots nevertheless. :-)

sarpba, 2024-07-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

syssi, 2024-07-08:
Some example frames for later analysis:

```
>>> 02:45:00:00:00:54:00:00:94:C6
<<< 02:45:00:00:00:54:00:55:00:10:0C:D1:0C:D0:0C:D0:0C:D0:0C:D0:0C:D1:0C:D0:0C:D0:0C:CF:0C:D0:0C:D0:0C:D0:0C:D0:0C:CF:0C:D0:0C:CD:0C:D1:0C:CD:14:7F:FF:FF:FE:2B:0B:E1:0B:E1:27:10:0E:8A:00:44:00:25:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:93:B6

>>> 03:45:00:00:00:54:00:00:55:0A
<<< 03:45:00:00:00:54:00:55:00:10:0C:CD:0C:D0:0C:CE:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CE:0C:D0:0C:CE:0C:D1:0C:CE:0C:D0:0C:D1:0C:CD:14:7E:FF:FF:FE:2D:0B:CD:0B:CD:27:10:0F:58:00:42:00:27:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:F8:07

>>> 02:45:00:00:00:54:00:00:94:C6
<<< 02:45:00:00:00:54:00:55:00:10:0C:D1:0C:CD:0C:D2:0C:CE:0C:D1:0C:CE:0C:D1:0C:CE:0C:D1:0C:CF:0C:D1:0C:CE:0C:D2:0C:CE:0C:D1:0C:CC:0C:D2:0C:CC:14:7F:FF:FF:FE:29:0B:E1:0B:E1:27:10:0E:89:00:44:00:25:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:24:CB

>>> 03:45:00:00:00:54:00:00:55:0A
<<< 03:45:00:00:00:54:00:55:00:10:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CD:0C:D0:0C:CE:0C:D0:0C:CD:0C:D1:0C:CD:0C:CF:0C:D1:0C:CD:14:7D:FF:FF:FE:28:0B:CD:0B:CD:27:10:0F:57:00:42:00:27:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:38:56

>>> 02:45:00:00:00:54:00:00:94:C6
<<< 02:45:00:00:00:54:00:55:00:10:0C:D1:0C:CE:0C:D1:0C:CE:0C:D1:0C:CF:0C:D1:0C:CE:0C:D1:0C:CE:0C:D1:0C:CE:0C:D1:0C:CE:0C:D1:0C:CB:0C:D1:0C:CB:14:7F:FF:FF:FE:2E:0B:E1:0B:E1:27:10:0E:88:00:44:00:25:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:9E:0C

>>> 03:45:00:00:00:54:00:00:55:0A
<<< 03:45:00:00:00:54:00:55:00:10:0C:CD:0C:D0:0C:CE:0C:D0:0C:CD:0C:D0:0C:CE:0C:D0:0C:CD:0C:D0:0C:CE:0C:D1:0C:CD:0C:D1:0C:CD:0C:D0:0C:D1:0C:CD:14:7E:FF:FF:FE:28:0B:CD:0B:CD:27:10:0F:56:00:42:00:27:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:EB:B7
```

syssi, 2024-07-08:
```
     0  02 device address
     1  45 function code 0x45 = custom function code
     2  00 start address high byte
     3  00 start address low byte
     4  00 number of registers to read high byte
     5  54 number of registers to read low byte
     6  00
     7  00
     8  94 CRC
     9  C6 CRC
```

syssi, 2024-07-08:
The request above retrieve 55 registers / 2*55 bytes from the BMS:

```
0x02        Device address
0x45        Custom function code
0x00 0x00   Start address
0x00 0x54   End address
0x00 0x55   Data length
0x00 0x10   16 cells
0x0C 0xD1   3281 = 3.281 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD1   3281 = 3.281 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xCF   3279 = 3.279 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xCF   3279 = 3.270 V
0x0C 0xD0   3280 = 3.280 V
0x0C 0xCD   3277 = 3.277 V
0x0C 0xD1   3281 = 3.281 V
0x0C 0xCD   3277 = 3.277 V
0x14 0x7F   5247 = 52.47 V total voltage
0xFF 0xFF
0xFE 0x2B   65067
0x0B 0xE1   3041 = 30.41 °C?
0x0B 0xE1   3041 = 30.41 °C?
0x27 0x10   10000
0x0E 0x8A   3722
0x00 0x44   68  SOC?
0x00 0x25   37  SOC?
0x00 0x64   100 SOH?
0x00 0x00
0x00 0x00
0x00 0x07    7
0x00 0x00
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0x00 0x00
0x00 0x00
0x00 
0x93 0xB6
```

syssi, 2024-07-08:
Please try to retrieve some data using this YAML from the RS485 port next to the CAN port:

```
substitutions:
  name: pace-bms
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  - id: bms1
    address: 0x02
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  - id: bms2
    address: 0x03
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

sensor:
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "bank1 cell count"
    address: 0
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 0

  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "bank2 cell count"
    address: 0
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: ""
    accuracy_decimals: 0

  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "bank3 cell count"
    address: 0
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: ""
    accuracy_decimals: 0
```

syssi, 2024-07-08:
It looks like the protocol isn't Modbus compatible. Let's query the data manually:

```
substitutions:
  name: pace-bms
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: true

interval:
  - interval: 15s
    then:
      - uart.write: [0x01, 0x45, 0x00, 0x00, 0x00, 0x54, 0x00, 0x00, 0xFF, 0xFF]
      - delay: 1s

      - uart.write: [0x02, 0x45, 0x00, 0x00, 0x00, 0x54, 0x00, 0x00, 0x94, 0xC6]
      - delay: 1s

      - uart.write: [0x03, 0x45, 0x00, 0x00, 0x00, 0x54, 0x00, 0x00, 0x55, 0x0A]
      - delay: 1s
```

sarpba, 2024-07-08:
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/esphome-web-51a298.yaml...
INFO Starting log output from 192.168.7.109 using esphome API
INFO Successfully connected to esphome-web-51a298 @ 192.168.7.109 in 0.101s
INFO Successful handshake with esphome-web-51a298 @ 192.168.7.109 in 0.072s
[17:25:14][I][app:100]: ESPHome version 2024.6.6 compiled on Jul  8 2024, 17:22:30
[17:25:14][I][app:102]: Project syssi.esphome-pace-bms version 1.0.0
[17:25:14][C][wifi:599]: WiFi:
[17:25:14][C][wifi:427]:   Local MAC: EC:DA:3B:51:A2:98
[17:25:14][C][wifi:432]:   SSID: [redacted]
[17:25:14][C][wifi:435]:   IP Address: 192.168.7.109
[17:25:14][C][wifi:439]:   BSSID: [redacted]
[17:25:14][C][wifi:440]:   Hostname: 'esphome-web-51a298'
[17:25:14][C][wifi:442]:   Signal strength: -48 dB ▂▄▆█
[17:25:14][C][wifi:446]:   Channel: 5
[17:25:14][C][wifi:447]:   Subnet: 255.255.255.0
[17:25:14][C][wifi:448]:   Gateway: 192.168.7.1
[17:25:14][C][wifi:449]:   DNS1: 192.168.7.1
[17:25:14][C][wifi:450]:   DNS2: 0.0.0.0
[17:25:14][C][logger:185]: Logger:
[17:25:14][C][logger:186]:   Level: DEBUG
[17:25:14][C][logger:188]:   Log Baud Rate: 115200
[17:25:14][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[17:25:14][C][uart.idf:139]: UART Bus 1:
[17:25:14][C][uart.idf:140]:   TX Pin: GPIO16
[17:25:14][C][uart.idf:141]:   RX Pin: GPIO17
[17:25:14][C][uart.idf:143]:   RX Buffer Size: 256
[17:25:14][C][uart.idf:145]:   Baud Rate: 9600 baud
[17:25:14][C][uart.idf:146]:   Data Bits: 8
[17:25:14][C][uart.idf:147]:   Parity: NONE
[17:25:14][C][uart.idf:148]:   Stop bits: 1
[17:25:15][C][mdns:115]: mDNS:
[17:25:15][C][mdns:116]:   Hostname: esphome-web-51a298
[17:25:15][C][esphome.ota:073]: Over-The-Air updates:
[17:25:15][C][esphome.ota:074]:   Address: esphome-web-51a298.local:3232
[17:25:15][C][esphome.ota:075]:   Version: 2
[17:25:15][C][safe_mode:018]: Safe Mode:
[17:25:15][C][safe_mode:020]:   Boot considered successful after 60 seconds
[17:25:15][C][safe_mode:021]:   Invoke after 10 boot attempts
[17:25:15][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[17:25:15][C][api:139]: API Server:
[17:25:15][C][api:140]:   Address: esphome-web-51a298.local:6053
[17:25:15][C][api:142]:   Using noise encryption: YES
[17:25:15][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:25:16][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:25:17][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:25:30][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:25:31][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:25:32][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:25:45][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:25:46][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:25:47][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:26:00][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:26:01][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:26:02][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:26:15][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:26:16][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:26:17][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:26:30][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:26:31][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:26:32][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:26:45][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:26:46][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:26:47][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:27:00][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:27:01][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:27:02][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:27:15][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:27:16][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:27:17][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:27:30][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:27:31][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:27:32][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:27:45][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:27:46][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:27:47][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:28:00][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:28:01][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:28:02][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:28:15][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:28:16][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:28:17][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:28:30][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3

syssi, 2024-07-08:
```
[17:25:15][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:25:16][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:25:17][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[17:25:30][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[17:25:31][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[17:25:32][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
```

Your BMS doesn't respond anymore. Did you attach the data cable properly?

sarpba, 2024-07-08:
Ok, I done a little test with offical program.
If I use the rs485 port near the can plug i can monitor only one battery.
if I use the plug rs485 port on gight side, I can monitor paralell the another 2 battery too, but slower. Directly conected battery respond time 0,5-1s, Paralell connected another two battery rebpond time around 2-15s.

sarpba, 2024-07-08:
Log from the right rs485 connector
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/esphome-web-51a298.yaml...
INFO Starting log output from 192.168.7.109 using esphome API
INFO Successfully connected to esphome-web-51a298 @ 192.168.7.109 in 0.036s
INFO Successful handshake with esphome-web-51a298 @ 192.168.7.109 in 0.073s
[18:23:24][I][app:100]: ESPHome version 2024.6.6 compiled on Jul  8 2024, 17:22:30
[18:23:24][I][app:102]: Project syssi.esphome-pace-bms version 1.0.0
[18:23:24][C][wifi:599]: WiFi:
[18:23:24][C][wifi:427]:   Local MAC: EC:DA:3B:51:A2:98
[18:23:24][C][wifi:432]:   SSID: [redacted]
[18:23:24][C][wifi:435]:   IP Address: 192.168.7.109
[18:23:24][C][wifi:439]:   BSSID: [redacted]
[18:23:24][C][wifi:440]:   Hostname: 'esphome-web-51a298'
[18:23:24][C][wifi:442]:   Signal strength: -56 dB ▂▄▆█
[18:23:24][C][wifi:446]:   Channel: 5
[18:23:24][C][wifi:447]:   Subnet: 255.255.255.0
[18:23:24][C][wifi:448]:   Gateway: 192.168.7.1
[18:23:24][C][wifi:449]:   DNS1: 192.168.7.1
[18:23:24][C][wifi:450]:   DNS2: 0.0.0.0
[18:23:24][C][logger:185]: Logger:
[18:23:24][C][logger:186]:   Level: DEBUG
[18:23:24][C][logger:188]:   Log Baud Rate: 115200
[18:23:24][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[18:23:24][C][uart.idf:139]: UART Bus 1:
[18:23:24][C][uart.idf:140]:   TX Pin: GPIO16
[18:23:24][C][uart.idf:141]:   RX Pin: GPIO17
[18:23:24][C][uart.idf:143]:   RX Buffer Size: 256
[18:23:24][C][uart.idf:145]:   Baud Rate: 9600 baud
[18:23:24][C][uart.idf:146]:   Data Bits: 8
[18:23:24][C][uart.idf:147]:   Parity: NONE
[18:23:24][C][uart.idf:148]:   Stop bits: 1
[18:23:24][C][mdns:115]: mDNS:
[18:23:24][C][mdns:116]:   Hostname: esphome-web-51a298
[18:23:24][C][esphome.ota:073]: Over-The-Air updates:
[18:23:24][C][esphome.ota:074]:   Address: esphome-web-51a298.local:3232
[18:23:24][C][esphome.ota:075]:   Version: 2
[18:23:24][C][safe_mode:018]: Safe Mode:
[18:23:24][C][safe_mode:020]:   Boot considered successful after 60 seconds
[18:23:24][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:23:24][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[18:23:24][C][api:139]: API Server:
[18:23:24][C][api:140]:   Address: esphome-web-51a298.local:6053
[18:23:24][C][api:142]:   Using noise encryption: YES
[18:23:24][D][uart_debug:114]: <<< 0F:45:00:00:00:54:00:00:55:5F
[18:23:24][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:00:94:C6
[18:23:24][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:55:00:10:0C:B9:0C:BA:0C:BB:0C:BC:0C:BB:0C:BD:0C:BB:0C:BB:0C:BB:0C:BB:0C:BA:0C:BA:0C:BA:0C:BB:0C:BA:0C:B8:0C:BD:0C:B8:14:5D:FF:FF:FA:EE:0B:D7:0B:D7:27:10:0A:EC:00:44:00:1B:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:C4:58
[18:23:25][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:00:55:0A
[18:23:25][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:55:00:10:0C:B7:0C:BD:0C:B5:0C:BC:0C:B4:0C:BC:0C:B5:0C:BC:0C:B5:0C:BC:0C:B5:0C:BE:0C:B2:0C:BF:0C:B4:0C:BD:0C:BF:0C:B2:14:5A:FF:FF:FB:46:0B:C3:0B:C3:27:10:0A:7C:00:42:00:1A:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:F7:5D
[18:23:25][D][uart_debug:114]: <<< 04:45:00:00:00:54:00:00:14:EC
[18:23:25][D][uart_debug:114]: <<< 05:45:00:00:00:54:00:00:D5:20
[18:23:26][D][uart_debug:114]: <<< 06:45:00:00:00:54:00:00:95:35
[18:23:26][D][uart_debug:114]: <<< 07:45:00:00:00:54:00:00:54:F9
[18:23:27][D][uart_debug:114]: <<< 08:45:00:00:00:54:00:00:14:B9
[18:23:27][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[18:23:27][D][uart_debug:114]: <<< 09:45:00:00:00:54:00:00:D5:75
[18:23:27][D][uart_debug:114]: <<< 0A:45:00:00:00:54:00:00:95:60
[18:23:28][D][uart_debug:114]: >>> 02:45:00:00:00:54:00:00:94:C6
[18:23:28][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:55:00:10:0C:B7:0C:BC:0C:B7:0C:BE:0C:B7:0C:BF:0C:B4:0C:BE:0B:45:00:00:00:54:00:00:54:AC:C8:63:BB:0C:BF:0C:B3:14:5B:FF:FF:FA:E2:0B:D7:0B:D7:27:10:0A:EB:00:44:00:1B:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:37:EC
[18:23:28][D][uart_debug:114]: <<< 0C:45:00:00:00:54:00:00:15:4A
[18:23:29][D][uart_debug:114]: <<< 0D:45:00:00:00:54:00:00:D4:86
[18:23:29][D][uart_debug:114]: >>> 03:45:00:00:00:54:00:00:55:0A
[18:23:29][D][uart_debug:114]: <<< 0E:45:00:00:00:54:00:00:94:93
[18:23:30][D][uart_debug:114]: <<< 0F:45:00:00:00:54:00:00:55:5F
[18:23:30][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:00:94:C6
[18:23:30][D][uart_debug:114]: <<< 02:45:00:00:00:54:00:55:00:10:0C:BE:0C:B6:0C:BB:0C:B9:0C:BD:0C:B9:0C:BD:0C:B9:0C:BB:0C:B7:0C:BB:0C:B7:0C:BE:0C:B8:0C:BD:0C:B6:0C:BE:0C:B6:14:5C:FF:FF:FA:DD:0B:D7:0B:D7:27:10:0A:EA:00:44:00:1B:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:F1:C2
[18:23:30][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:00:55:0A
[18:23:31][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:55:00:10:0C:BE:0C:B5:0C:BC:0C:B4:0C:BE:0C:B3:0C:BD:0C:B5:0C:BC:0C:B5:0C:BC:0C:B7:0C:BB:0C:B7:0C:BD:0C:B5:0C:BE:0C:B3:14:5B:FF:FF:FB:19:0B:C3:0B:C3:27:10:0A:79:00:42:00:1A:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:7F:4C
[18:23:31][D][uart_debug:114]: <<< 04:45:00:00:00:54:00:00:14:EC
[18:23:31][D][uart_debug:114]: <<< 05:45:00:00:00:54:00:00:D5:20
[... 122 lines cut ...]
[18:24:11][D][uart_debug:114]: <<< 06:45:00:00:00:54:00:00:95:35
[18:24:11][D][uart_debug:114]: <<< 07:45:00:00:00:54:00:00:54:F9
[18:24:12][D][uart_debug:114]: >>> 01:45:00:00:00:54:00:00:D4:D3
[... cut ...]
[18:24:16][D][uart_debug:114]: <<< 03:45:00:00:00:54:00:55:00:10:0C:BD:0C:B5:0C:BC:0C:B3:0C:BD:0C:B4:0C:BC:0C:B3:0C:BC:0C:B2:0C:BC:0C:B7:0C:BB:0C:B7:0C:BD:0C:B5:0C:BD:0C:B2:14:5A:FF:FF:FB:20:0B:C3:0B:C3:27:10:0A:6A:00:42:00:1A:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:84:73
[18:24:16][D][uart_debug:114]: <<< 04:45:00:00:00:54:00:00:14:EC
[18:24:16][D][uart_debug:114]: <<< 05:45:00:00:00:54:00:00:D5:20
[18:24:17][D][uart_debug:114]: <<< 06:45:00:00:00:54:00:00:95:35
[18:24:17][D][uart_debug:114]: <<< 07:45:00:00:00:54:00:00:54:F9
[18:24:18][D][uart_debug:114]: <<< 08:45:00:00:00:54:00:00:14:B9

sarpba, 2024-07-08:
My modbus "hardver" for PC connection is a simple yp-05, and a cheap HW-519 modbus modul from ali (I use lots of them). I use this modbus modul for esp32 too. Maybe my cable long? I use 2-2,5m cat5e cable on rs485 side.

syssi, 2024-07-08:
On the right RS485 all systems are responding to our requests:

```
# Request cell infos from device 0x01
>>> 01:45:00:00:00:54:00:00:D4:D3
# Device 0x01 responds
<<< 01:45:00:00:00:54:00:55:00:10:0C:BA:0C:B4:0C:B9:0C:B1:0C:B9:0C:B2:0C:B9:0C:B4:0C:B8:0C:B3:0C:B8:0C:B1:0C:B9:0C:B5:0C:B8:0C:B2:0C:BA:0C:B1:14:56:FF:FF:FA:86:0B:E1:0B:D7:29:76:0C:F1:00:47:00:1F:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:58:B8:08:45:00:00:00:54:00:00:14:B9

# Request cell infos from device 0x02
>>> 02:45:00:00:00:54:00:00:94:C6
# Device 0x02 responds
<<< 02:45:00:00:00:54:00:55:00:10:0C:BC:0C:B4:0C:BD:0C:B5:0C:BE:0C:B5:0C:BD:0C:B6:0C:BC:0C:B4:0C:BC:0C:B4:0C:BE:0C:B3:0C:BD:0C:B2:0C:BE:0C:B2:14:5A:FF:FF:FA:CA:0B:D7:0B:D7:27:10:0A:D9:00:44:00:1B:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:2C:DD

# Request cell infos from device 0x03
>>> 03:45:00:00:00:54:00:00:55:0A
# Device 0x03 responds
<<< 03:45:00:00:00:54:00:55:00:10:0C:BE:0C:B3:0C:BC:0C:B4:0C:BE:0C:B3:0C:BC:0C:B4:0C:BC:0C:B2:0C:BC:0C:B7:0C:BB:0C:B7:0C:BD:0C:B4:0C:BE:0C:B2:14:5A:FF:FF:FB:14:0B:C3:0B:C3:27:10:0A:6A:00:42:00:1A:00:64:00:00:00:00:00:07:00:00:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:00:00:00:00:00:87:2D:0D:45:00:00:00:54:00:00:D4:86
```

But you are **ignoring one important detail** at the moment:

If this is a Modbus protocol there is just **one** master allowed. At the moment BMS1 (0x01) acts as master and we aren't allowed to query data using another master (the ESP!). 

This is the reason why I would prefer to use the RS485 port next to the CAN bus.

sarpba, 2024-07-08:
It's true.

An absolutely suitable solution for me is to read out the data one by one next to the can port. In the worst case, you need 3 esp and 3 modbus modules.

syssi, 2024-07-08:
In best case you can connect a single ESP + single RS485 converter to all RS485 ports and the BMS just responds to the dedicated device address (hopefully!).

syssi, 2024-07-08:
Could you try the YAML configuration with the manual requests another time on the RS485 port next to the CAN bus?

```
      - uart.write: [0x01, 0x45, 0x00, 0x00, 0x00, 0x54, 0x00, 0x00, 0xD4, 0xD3]
      - delay: 1s

      - uart.write: [0x02, 0x45, 0x00, 0x00, 0x00, 0x54, 0x00, 0x00, 0x94, 0xC6]
      - delay: 1s

      - uart.write: [0x03, 0x45, 0x00, 0x00, 0x00, 0x54, 0x00, 0x00, 0x55, 0x0A]
      - delay: 1s
```

Still silence?

sarpba, 2024-07-08:
yes, unfortunatly, sitill silence

What do you think, does it make sense to extract data from PC serial communication with a serial port analyzer? (Reverse enginering from offical softver communication)

syssi, 2024-07-08:
We can use an ESP to sniff the traffic:

1. Connect the PC using an RS485 converter to the RS485 port
2. Attach a RS485 converter module to the ESP (as always) and connect A+/B- to the BMS too (some lines/pins as the PC dongle)
3. Use this YAML again to dump the data: https://github.com/syssi/esphome-pace-bms/issues/37#issuecomment-2212494967

sarpba, 2024-07-08:
SOC low alarm on 2 and 3 battrey
[SOC_LOW_Alarm_RealTimeData_20240708192501.csv](https://github.com/user-attachments/files/16132543/SOC_LOW_Alarm_RealTimeData_20240708192501.csv)

syssi, 2024-07-08:
These are Seplos ASCII frames (protocol version 2.2):

```
~22014A00E0C60010A1146D100CC90CC20CC60CC40CC50CC30CC60CC30CC50CC30CC50CC30CC50CC40CC50CC301540136017C0401400136012C0136F9400000006401297611A60047000000020000000000230000000000000000000000000000000000000000000000D4AB
~22014A00E0C60010A01467100CC90CBA0CC90CB70CC90CBA0CC90CBA0CC90CB90CC90CB90CC90CB60CC90CB801540136017C0401400136012C0136F93C0000006401297611A60047000000020000000000230000000000000000000000000000000000000000000000D454
~22014A00E0C60010A0146D100CC50CC30CC70CC30CC70CC20CC70CC30CC70CC30CC60CC30CC70CC30CC60CC30154013601720401400136012C0136F9510000006401297611A50047000000020000000000230000000000000000000000000000000000000000000000D4B8
~22014A00E0C60010A01468100CB40CC90CBC0CC90CB90CC90CBC0CC90CB90CC90CBC0CC90CBB0CC90CBA0CC80154012C01720401360136012C0136F9430000006401297611A50047000000020000000000230000000000000000000000000000000000000000000000D44C
```

sarpba, 2024-07-08:
Here is the snifed datas:

[bat1.txt](https://github.com/user-attachments/files/16133352/bat1.txt)
[batery_1_ from_offical_ RealTimeData_20240708202449.csv](https://github.com/user-attachments/files/16133376/batery_1_.from_offical_.RealTimeData_20240708202449.csv)

[bat2.txt](https://github.com/user-attachments/files/16133353/bat2.txt)
[batery_2_from_offical_RealTimeData_20240708203247.csv](https://github.com/user-attachments/files/16133378/batery_2_from_offical_RealTimeData_20240708203247.csv)

[bat3.txt](https://github.com/user-attachments/files/16133354/bat3.txt)
[batery_3_from_offical_RealTimeData_20240708204012.csv](https://github.com/user-attachments/files/16133380/batery_3_from_offical_RealTimeData_20240708204012.csv)

I tryed read another battery (just for adresses)
[bat1_on_bat3_connector.txt](https://github.com/user-attachments/files/16133370/bat1_on_bat3_connector.txt)
[bat2_on_bat1_connector.txt](https://github.com/user-attachments/files/16133371/bat2_on_bat1_connector.txt)
[bat3_on_bat1_connector.txt](https://github.com/user-attachments/files/16133372/bat3_on_bat1_connector.txt)

syssi, 2024-07-08:
```
# bat1.txt
>>> 7E:32:32:30:31:34:41:34:32:45:30:30:32:30:31:46:44:32:38:0D
<<< 7E:32:32:30:31:34:41:30:30:45:30:43:36:30:30:30:35:33:44:31:34:30:37:31:30:30:43:38:34:30:43:38:35:30:43:38:35:30:43:38:34:30:43:38:35:30:43:38:35:30:43:38:35:30:43:38:35:30:43:38:35:30:43:38:35:30:43:38:34:30:43:38:34:30:43:38:35:30:43:38:35:30:43:38:34:30:43:38:34:30:31:34:41:30:31:33:36:30:31:33:36:30:34:30:31:33:36:30:31:34:30:30:31:34:30:30:31:33:36:30:30:30:30:30:30:30:30:30:30:36:34:30:31:32:39:37:36:30:35:38:46:30:30:34:37:30:30:30:30:30:30:30:30:30:30:30:30:30:30:38:30:30:30:32:33:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:44:35:38:42:0D

# hex2ascii
>>> ~22014A42E00201FD28
<<< ~22014A00E0C600053D1407100C840C850C850C840C850C850C850C850C850C850C840C840C850C850C840C84014A01360136040136014001400136000000000064012976058F0047000000000000008000230000000000000000000000000000000000000000000000D58B
``` 

```
# bat2.txt
>>> 7E:32:32:30:32:34:41:34:32:45:30:30:32:30:32:46:44:32:36:0D
<<< 7E:32:32:30:32:34:41:30:30:45:30:43:36:30:30:30:33:39:38:31:34:30:41:31:30:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:37:30:43:38:37:30:43:38:37:30:43:38:37:30:43:38:37:30:43:38:36:30:43:38:37:30:43:38:36:30:43:38:37:30:43:38:36:30:43:38:36:30:43:38:34:30:31:34:41:30:31:33:36:30:31:33:36:30:34:30:31:33:36:30:31:33:36:30:31:33:36:30:31:33:36:30:30:30:30:30:30:30:30:30:30:36:34:30:31:32:37:31:30:30:33:39:38:30:30:34:34:30:30:30:30:30:30:30:30:30:30:30:30:30:30:38:30:30:30:32:33:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:44:35:38:33:0D

# hex2ascii
>>> ~22024A42E00202FD26
<<< ~22024A00E0C6000398140A100C860C860C860C860C870C870C870C870C870C860C870C860C870C860C860C84014A0136013604013601360136013600000000006401271003980044000000000000008000230000000000000000000000000000000000000000000000D583
```

```
# bat3.txt
>>> 7E:32:32:30:33:34:41:34:32:45:30:30:32:30:33:46:44:32:34:0D
<<< 7E:32:32:30:33:34:41:30:30:45:30:43:36:30:30:30:33:34:35:31:34:30:39:31:30:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:35:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:36:30:43:38:37:30:43:38:35:30:43:38:37:30:43:38:36:30:43:38:35:30:31:33:36:30:31:32:32:30:31:32:43:30:34:30:31:32:32:30:31:32:32:30:31:32:32:30:31:32:32:30:30:30:30:30:30:30:30:30:30:36:34:30:31:32:37:31:30:30:33:34:35:30:30:34:33:30:30:30:30:30:30:30:30:30:30:30:30:30:30:38:30:30:30:32:33:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:44:35:42:41:0D

# hex2ascii
~22034A42E00203FD24
~22034A00E0C60003451409100C860C860C860C860C860C850C860C860C860C860C860C870C850C870C860C8501360122012C04012201220122012200000000006401271003450043000000000000008000230000000000000000000000000000000000000000000000D5BA
```

```
# bat1_on_bat3_connector.txt
<<< 7E:32:32:30:31:34:41:34:32:45:30:30:32:30:31:46:44:32:38:0D
<<< 7E:32:32:30:31:34:41:35:31:30:30:30:30:46:44:41:30:0D
<<< 7E:32:32:30:31:34:41:42:30:36:30:30:41:30:31:30:31:30:33:46:46:30:30:46:42:36:43:0D
<<< 7E:32:32:30:31:34:41:34:44:30:30:30:30:46:44:38:45:0D
<<< 7E:32:32:30:31:34:41:34:32:45:30:30:32:30:31:46:44:32:38:0D

# hex2ascii
~22014A42E00201FD28
~22014A510000FDA0
~22014AB0600A010103FF00FB6C
~22014A4D0000FD8E
~22014A42E00201FD28
```

```
# bat2_on_bat1_connector.txt
<<< 7E:32:32:30:32:34:41:35:31:30:30:30:30:46:44:39:46:0D
<<< 7E:32:32:30:32:34:41:42:30:36:30:30:41:30:32:30:31:30:33:46:46:30:30:46:42:36:41:0D
<<< 7E:32:32:30:32:34:41:34:44:30:30:30:30:46:44:38:44:0D
<<< 7E:32:32:30:32:34:41:34:32:45:30:30:32:30:32:46:44:32:36:0D
<<< 7E:32:32:30:32:34:41:34:32:45:30:30:32:30:32:46:44:32:36:0D
<<< 7E:32:32:30:32:34:41:34:32:45:30:30:32:30:32:46:44:32:36:0D
<<< 7E:32:32:30:32:34:41:34:32:45:30:30:32:30:32:46:44:32:36:0D

# hex2ascii
~22024A510000FD9F
~22024AB0600A020103FF00FB6A
~22024A4D0000FD8D
~22024A42E00202FD26
~22024A42E00202FD26
~22024A42E00202FD26
~22024A42E00202FD26
```

```
# bat3_on_bat1_connector.txt
<<< 7E:32:32:30:33:34:41:34:32:45:30:30:32:30:33:46:44:32:34:0D
<<< 7E:32:32:30:33:34:41:35:31:30:30:30:30:46:44:39:45:0D
<<< 7E:32:32:30:33:34:41:42:30:36:30:30:41:30:33:30:31:30:33:46:46:30:30:46:42:36:38:0D
<<< 7E:32:32:30:33:34:41:34:44:30:30:30:30:46:44:38:43:0D
<<< 7E:32:32:30:33:34:41:34:32:45:30:30:32:30:33:46:44:32:34:0D
<<< 7E:32:32:30:33:34:41:34:32:45:30:30:32:30:33:46:44:32:34:0D
<<< 7E:32:32:30:33:34:41:34:32:45:30:30:32:30:33:46:44:32:34:0D

# hex2ascii 
~22034A42E00203FD24
~22034A510000FD9E
~22034AB0600A030103FF00FB68
~22034A4D0000FD8C
~22034A42E00203FD24
~22034A42E00203FD24
~22034A42E00203FD24
```

sarpba, 2024-07-09:
I experimented a bit with the battery. I used the following program:
https://github.com/ClassicDIY/ModbusTool

If i press read holding register, button the battery is answer:

[14:02:40][D][uart_debug:114]: <<< 01:03:00:00:00:40:44:3A
[14:02:40][D][uart_debug:114]: <<< 01:03:00:40:00:00:00:00:01:00:01:00:00:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:10:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:21:00:00:00:00:00:00:00:00:00:04:0B:CC:0B:CC:0B:CC:0B:CC:00:00:00:00:00:00:00:00:00:00:00:00:00:32:00:00:02:12:00:22:00:01:9E:9C:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:50:97

sarpba, 2024-07-09:
> The request above retrieve 55 registers / 2*55 bytes from the BMS:
> 
> ```
> Addr  Payload   Description
> 
>       0x02        Device address
>       0x45        Custom function code
>       0x00 0x00   Start address
>       0x00 0x54   End address
>       0x00 0x55   Data length (0x55 = 85 bytes)
>   0   0x10   16 cells
>   1   0x0C 0xD1   3281 = 3.281 V
>   3   0x0C 0xD0   3280 = 3.280 V
>   5   0x0C 0xD0   3280 = 3.280 V
>   7   0x0C 0xD0   3280 = 3.280 V
>   9   0x0C 0xD0   3280 = 3.280 V
>  11   0x0C 0xD1   3281 = 3.281 V
>  13   0x0C 0xD0   3280 = 3.280 V
>  15   0x0C 0xD0   3280 = 3.280 V
>  17   0x0C 0xCF   3279 = 3.279 V
>  19   0x0C 0xD0   3280 = 3.280 V
>  21   0x0C 0xD0   3280 = 3.280 V
>  23   0x0C 0xD0   3280 = 3.280 V
>  25   0x0C 0xD0   3280 = 3.280 V
>  27   0x0C 0xCF   3279 = 3.270 V
>  29   0x0C 0xD0   3280 = 3.280 V
>  31   0x0C 0xCD   3277 = 3.277 V
>  33   0x0C 0xD1   3281 = 3.281 V
>  35   0x0C 0xCD   3277 = 3.277 V
>  37   0x14 0x7F   5247 = 52.47 V total voltage
>  39   0xFF 0xFF
>  41   0xFE 0x2B   65067
>  43   0x0B 0xE1   3041 = 30.41 °C?
>  45   0x0B 0xE1   3041 = 30.41 °C?
>  47   0x27 0x10   10000
>  49   0x0E 0x8A   3722
>  51   0x00 0x44   68  SOC?
>  53   0x00 0x25   37  SOC?
>  55   0x00 0x64   100 SOH?
>  57   0x00 0x00
>  59   0x00 0x00
>  61   0x00 0x07    7
>  63   0x00 0x00
>  65   0xFF 0xFF
>  67   0xFF 0xFF
>  69   0xFF 0xFF
>  71   0xFF 0xFF
>  73   0xFF 0xFF
>  75   0xFF 0xFF
>  77   0xFF 0xFF
>  79   0x00 0x00
>  81   0x00 0x00
>  83   0x00 
>  84   0x93 0xB6
> ```

43 temperature
45 temperature
51 chargeing cycles
53 SOC
55 SOH

sarpba, 2024-07-09:
Not 100%, but I got a tip from the reseller. (Pylontech)

sarpba, 2024-07-09:
With a modified slepos esp32-example.yaml
substitutions:
  name: esphome-web-51a298
  friendly_name: bms
  device_description: "Monitor a Seplos BMS via RS485"
  external_components_source: github://syssi/esphome-seplos-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-seplos-bms"
    version: 1.1.0

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  baud_rate: 0 #disable logging over uart

api:
  reboot_timeout: 10min
  encryption:
    key: "PkyF/o6kVkQDEbltsQMMEuVk9wII7LKYz0qxXhEwmC5="

uart:
  id: uart_0
  # Please set the default baudrate of your Seplos BMS model here. It's sometimes 19200 baud instead of 9600.
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  # The increased RX buffer size is important because
  # the full BMS response must fit into the buffer
  rx_buffer_size: 384
  debug:
    direction: BOTH
    dummy_receiver: false

seplos_modbus:
  id: modbus0
  uart_id: uart_0
  rx_timeout: 150ms

seplos_bms:
  id: bms0
  # Dip switch configuration of a single pack setup / address 0x0
  #  8    7    6    5    4    3    2    1
  # off, off, off, off, off, off, off, off
  address: 0x01
  # Known protocol versions: 0x20 (Seplos), 0x26 (Boqiang)
  protocol_version: 0x22
  seplos_modbus_id: modbus0
[... 71 lines cut ...]
      name: "${name} residual capacity"
    battery_capacity:
      name: "${name} battery capacity"
    rated_capacity:
      name: "${name} rated capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_cycles:
      name: "${name} charging cycles"
    state_of_health:
      name: "${name} state of health"
    port_voltage:
      name: "${name} port voltage"

some kind of communication started (adress 0x01, protocoll 0x22):

7E:32:32:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:33:0D
[01:56:02][D][uart_debug:114]: <<< 7E:32:32:30:31:34:41:30:30:45:30:43:36:30:30:30:35:37:39:31:33:46:41:31:30:30:43:37:43:30:43:37:45:30:43:37:45:30:43:37:43:30:43:37:44:30:43:37:44:30:43:37:44:30:43:37:44:30:43:37:43:30:43:37:44:30:43:37:43:30:43:37:43:30:43:37:44:30:43:37:44:30:43:37:43:30:43:37:43:30:31:34:30:30:31:32:43:30:31:32:43:30:34:30:31:32:43:30:31:32:43:30:31:32:43:30:31:32:43:30:30:30:30:30:30:30:30:30:30:36:34:30:31:32:39:37:36:30:35:43:46:30:30:34:38:30:30:30:30:30
[01:56:02][W][seplos_bms:022]: Unhandled data received (data_len: 0xC6): 22.01.4A.00.E0.C6.00.05.79.13.FA.10.0C.7C.0C.7E.0C.7E.0C.7C.0C.7D.0C.7D.0C.7D.0C.7D.0C.7C.0C.7D.0C.7C.0C.7C.0C.7D.0C.7D.0C.7C.0C.7C.01.40.01.2C.01.2C.04.01.2C.01.2C.01.2C.01.2C.00.00.00.00.00.64.01.29.76.05.CF.00.48.00.00.00.00.00.00.00.80.00.23.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (105)
[01:56:02][D][uart_debug:114]: <<< 30:30:30:30:30:30:30:30:30:38:30:30:30:32:33:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:44:34:34:35:0D

[01:56:12][D][uart_debug:114]: >>> 7E:32:32:30:31:34:36:34:32:45:30:30:32:30:31:46:44:33:33:0D
[01:56:12][D][uart_debug:114]: <<< 7E:32:32:30:31:34:41:30:30:45:30:43:36:30:30:30:35:37:39:31:33:46:41:31:30:30:43:37:43:30:43:37:44:30:43:37:44:30:43:37:43:30:43:37:44:30:43:37:43:30:43:37:44:30:43:37:44:30:43:37:43:30:43:37:44:30:43:37:43:30:43:37:43:30:43:37:44:30:43:37:44:30:43:37:43:30:43:37:43:30:31:34:30:30:31:32:43:30:31:32:43:30:34:30:31:32:43:30:31:32:43:30:31:32:43:30:31:32:43:30:30:30:30:30:30:30:30:30:30:36:34:30:31:32:39:37:36:30:35:43:46:30:30:34:38:30:30:30:30:30
[01:56:12][W][seplos_bms:022]: Unhandled data received (data_len: 0xC6): 22.01.4A.00.E0.C6.00.05.79.13.FA.10.0C.7C.0C.7D.0C.7D.0C.7C.0C.7D.0C.7C.0C.7D.0C.7D.0C.7C.0C.7D.0C.7C.0C.7C.0C.7D.0C.7D.0C.7C.0C.7C.01.40.01.2C.01.2C.04.01.2C.01.2C.01.2C.01.2C.00.00.00.00.00.64.01.29.76.05.CF.00.48.00.00.00.00.00.00.00.80.00.23.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (105)
[01:56:12][D][uart_debug:114]: <<< 30:30:30:30:30:30:30:30:30:38:30:30:30:32:33:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:30:44:34:34:38:0D

syssi, 2024-07-10:
I did the same already. The Seplos component isn't compatible.

sarpba, 2024-07-10:
It's growing slowly, we need those crazy registers, but give the data back to the homeassistant already, the data is just not in the right place. :)
I modified slepos_bms.ccp a little.
https://github.com/sarpba/esphome-seplos-bms/blob/main/components/seplos_bms/seplos_bms.cpp
[image]

sarpba, 2024-07-10:
This is how I stand now. It takes time to find out 1-2 things.
[image]

syssi, 2024-07-11:
Good job! Let's summarize our findings:

1. The right RS485 ports are used the chain the battery packs. The first battery pack acts as mater (`0x01`) and queries up to 15 slaves (0x02, 0x03, ..., 0x0f) periodically. The protocol looks like Modbus but a custom function code (`0x45`) is used.
2. The right RS485 ports can be used to sniff data of the slaves. If some data from the master (`0x01`) is requested it does respond but a second masters on the same bus isn't a good idea. Transmissions on the right RS485 ports should be avoided therefore.
3. The left ports doesn't respond to requests with the custom function code (`0x45`). The configuration tool of the manufacturer uses a Seplos/Pylontech like protocol:
```
"~2203 ... \n"
 | | |
 | | Device address
 | Protocol version 2.2
 ^ Preamble / Start of frame
```

TODOs:

Compare the different `CID2` bytes with a Pylontech implementation. Does the BMS supports the same CID2s/commands as a Pylontech battery pack?

syssi, 2024-07-16:
Let's move this discussion to the Seplos BMS project because of the Seplos like protocol.