syssi/esphome-pace-bms #39: Got my hands on a Revov R100 running pace bms p16s100a-12720-4.05G

JacquesdeBruyn, 2024-07-15:
Not getting anything from the inverter rs485 port on the revov using the code from the katbatt. i have changed the inverter protocol to pylon f to match the katbatt. also having them in series the 232 port can read both. Still playing around with adress etc to see if I can find a working match.

syssi, 2024-07-16:
Could you describe your setup in more detail? What have you already tried? Which options/protocols are available at the BMS tool? Are you sure you have wired the RS485/RS232 converters correctly?

JacquesdeBruyn, 2024-07-16:
I already have a pace battery. The top one on your working list. I just moved the rj45 from the rs485 on that to the revov. So far I've adjusted the addresses around nd changed the protocol to be the same as the working battery. No response to either. The battery comms side does have traffic. So not sure if there's a special code to read second battery soc through the first.

JacquesdeBruyn, 2024-07-16:
If I recall I started your code down the pace bms route.

syssi, 2024-07-22:
Hmmm..  could you try to compare the BMS model number and firmware version of both battery banks? Is there a difference?

syssi, 2024-07-22:
`P16S200A-21473-1.03A310771030800025A` (working) vs. `p16s100a-12720-4.05G`

JacquesdeBruyn, 2024-07-22:
I didn't check firmware version. Also borrowed a cable to change settings. I'll see if I have a connector for it somewhere. I already have a usb to ttl board ch340 if I'm not mistaken. Same chip the cable I borrowed had. And I haven't found specific specs on this bms yet. But I'll see if I can dig some up somewhere.

JacquesdeBruyn, 2024-07-22:
I also think I can try shifting around between the different rs485 pin sets. Also try reversing rx and tx. Just maybe that gets me a response at least. Will update when I get a chance to mess around

JacquesdeBruyn, 2024-07-29:
With the little information available online it seems reading this bms through the rs232 is the only option. I'll see if i can find a converter as this seems to be different from the standard ch340 ttl output

JacquesdeBruyn, 2024-07-29:
Unless there's an option to pull the data through the already working bms?

JacquesdeBruyn, 2024-07-29:
All I need is power in and out and soc. Just so I can properly log everything

syssi, 2024-07-29:
Are you aware of my `esphome-seplos-bms` project? This is probably compatible with the `PYLON` protocol on the RS232 port.

JacquesdeBruyn, 2024-07-29:
Pace bms is way better in my opinion though. That bms counts upwards of 10 cycles daily on my solar....

syssi, 2024-07-29:
It looks like there are some PACE BMS firmwares which doesn't support Modbus via RS232.

See https://github.com/syssi/esphome-pace-bms/issues/32

If you cannot convince your Pace BMS to speak Modbus you could try the Pylon protocol using the Seplos project.

JacquesdeBruyn, 2024-07-29:
I'm trying to get it through the rs485. Simply refusing to give any response. I do get data on the parallel485 ports. Can access data from both batteries is I connect the parallel ports and access one through the 232. Either way I'll have to order some ttl to 232 converters. So it'll take some time

JacquesdeBruyn, 2024-07-30:
[image]

JacquesdeBruyn, 2024-07-30:
Got modbus out. For r100 revov, set can protocol to blank and rs485 to growatt. I'm still verifying that everything is working properly but it's giving me proper output so far. 😁

JacquesdeBruyn, 2024-07-30:
[11:13:23][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[11:13:23][D][modbus:125]: Modbus error function code: 0x83 exception: 2
[11:13:23][E][modbus_controller:088]: Modbus error function code: 0x3 exception: 2 
[11:13:23][E][modbus_controller:097]: Modbus error - last command: function code=0x3  register address = 0x0  registers count=8 payload size=0
[11:13:23][D][uart_debug:114]: <<< 01:83:02:C0:F1
[11:13:23][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[11:13:23][D][text_sensor:064]: 'pace-bms warnings': Sending state ''
[11:13:23][D][text_sensor:064]: 'pace-bms protections': Sending state ''
[11:13:23][D][text_sensor:064]: 'pace-bms errors': Sending state ''
[11:13:23][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[11:13:23][D][sensor:094]: 'pace-bms balance status': Sending state 0.00000  with 0 decimals of accuracy
[11:13:23][D][uart_debug:114]: <<< 01:03:08:00:00:00:00:00:00:00:00:95:D7
[11:13:23][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 1': Sending state 0.00000 V with 3 decimals of accuracy
[11:13:24][D][:457]: Cell voltage 1: 0.000 V
[11:13:24][D][:457]: Cell voltage 2: 1.544 V
[11:13:24][D][:457]: Cell voltage 3: 2.991 V
[11:13:24][D][:457]: Cell voltage 4: 25.083 V
[11:13:24][D][:457]: Cell voltage 5: 0.102 V
[11:13:24][D][:457]: Cell voltage 6: 0.000 V
[11:13:24][D][:457]: Cell voltage 7: 0.074 V
[11:13:24][D][:457]: Cell voltage 8: 5.440 V
[11:13:24][D][:457]: Cell voltage 9: 1.544 V
[11:13:24][D][:457]: Cell voltage 10: 0.015 V
[11:13:24][D][:457]: Cell voltage 11: 10.000 V
[11:13:24][D][:457]: Cell voltage 12: 7.453 V
[11:13:24][D][:457]: Cell voltage 13: 10.000 V
[11:13:24][D][:457]: Cell voltage 14: 0.257 V
[11:13:24][D][:457]: Cell voltage 15: 0.002 V
[11:13:24][D][:457]: Cell voltage 16: 0.000 V
[11:13:24][D][sensor:094]: 'pace-bms average cell voltage': Sending state 4.03156 V with 3 decimals of accuracy
[11:13:24][D][sensor:094]: 'pace-bms min cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[11:13:24][D][sensor:094]: 'pace-bms max cell voltage': Sending state 25.08300 V with 3 decimals of accuracy
[11:13:24][D][sensor:094]: 'pace-bms min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[11:13:24][D][sensor:094]: 'pace-bms max voltage cell': Sending state 4.00000  with 0 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 25.08
[11:13:24][D][sensor:094]: 'pace-bms delta cell voltage': Sending state 25.08300 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 1544.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 2': Sending state 1.54400 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 2991.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 3': Sending state 2.99100 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 25083.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 4': Sending state 25.08300 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 102.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 5': Sending state 0.10200 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 6': Sending state 0.00000 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 74.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 7': Sending state 0.07400 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 5440.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 8': Sending state 5.44000 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 1544.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 9': Sending state 1.54400 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 15.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 10': Sending state 0.01500 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 10000.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 11': Sending state 10.00000 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 7453.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 12': Sending state 7.45300 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 10000.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 13': Sending state 10.00000 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 257.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 14': Sending state 0.25700 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 15': Sending state 0.00200 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[11:13:24][D][sensor:094]: 'pace-bms cell voltage 16': Sending state 0.00000 V with 3 decimals of accuracy
[11:13:24][D][modbus_controller.sensor:025]: Sensor new state: 256.00
[11:13:24][D][sensor:094]: 'pace-bms battery temperature 1': Sending state 25.60000 ˚C with 1 decimals of accuracy
[... 117 lines cut ...]
[... cut ...]
[11:13:33][D][text_sensor:064]: 'pace-bms errors': Sending state ''
[11:13:33][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[11:13:33][D][sensor:094]: 'pace-bms balance status': Sending state 0.00000  with 0 decimals of accuracy
[11:13:33][D][uart_debug:114]: <<< 01:03:08:00:00:00:00:00:00:00:00:95:D7
[11:13:33][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[11:13:34][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[11:13:34][D][sensor:094]: 'pace-bms cell voltage 1': Sending state 0.00000 V with 3 decimals of accuracy

JacquesdeBruyn, 2024-07-30:
Values are complete garbage. I'll try different protocols later. have to run

syssi, 2024-07-30:
Correct. You can instantiate multiple controllers on the same bus with different slave addresses assigned. Ping me if you need an example how to do it.

JacquesdeBruyn, 2024-07-30:
I think I saw something like that somewhere here. Basically just copy and paste everything from the address selection down, just changing the address. Correct?

syssi, 2024-07-30:
Feel free to post your draft/YAML.

syssi, 2024-07-30:
It should look like this:

```
substitutions:
  name: pace-bms
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

# api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  - id: bms1
    address: 0x02
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

sensor:
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} total voltage"
    address: 1
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 2
    filters:
      - multiply: 0.01

  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name} total voltage"
    address: 1
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 2
    filters:
      - multiply: 0.01
```

JacquesdeBruyn, 2024-07-31:
For anyone interested Revov R100 "P16S100A-12720-4.05G" working with example code using RS485 protocol "SRNE" some others gave most readings but soc and current weren't reporting properly.

I will post my working code monitoring 2 different batteries bms0 being P16S200A-21473-1.03A310771030800025A and bms1 being P16S100A-12720-4.05G
Both running PACE_MODBUS

JacquesdeBruyn, 2024-07-31:
substitutions:
  name: pace-bms
  name0: Katbatt
  name1: Revov
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
 
 

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "VWCTZnKZwlLn5Z5vNwdvPk0+51EFx2radcwh9g0BbJY="

ota:
  platform: esphome

logger:
  level: DEBUG

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    flow_control_pin: 18
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  - id: bms1
    # Slave address 0x02
    address: 0x02
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

sensor:

  # 0x0000  Current                               2 byte   R   int16  10mA (Positive: chargingm Negative: discharging)
[... 3035 lines cut ...]
    skip_updates: 60
    address: 150
    register_type: holding
    register_count: 10
    response_size: 20

  # 160  Model SN                             20 byte  RW  uint16  ASCII BMS Manufacturer
  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name1} bms serial number"
    skip_updates: 60
    address: 160
    register_type: holding
    register_count: 10
    response_size: 20

  # 170  PACK SN                              20 byte  RW  uint16  ASCII PACK Manufacturer
  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name1} battery pack serial number"
    skip_updates: 60
    address: 170
    register_type: holding
    register_count: 10
    response_size: 20

JacquesdeBruyn, 2024-07-31:
`substitutions:
  name: pace-bms
  name0: Katbatt
  name1: Revov
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-pace-bms"
    version: 1.0.0

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
 
 

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "VWCTZnKZwlLn5Z5vNwdvPk0+51EFx2radcwh9g0BbJY="

ota:
  platform: esphome

logger:
  level: DEBUG

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    flow_control_pin: 18
    uart_id: uart_0
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  - id: bms1
    # Slave address 0x02
    address: 0x02
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s

sensor:

  # 0x0000  Current                               2 byte   R   int16  10mA (Positive: chargingm Negative: discharging)
[... 3035 lines cut ...]
    skip_updates: 60
    address: 150
    register_type: holding
    register_count: 10
    response_size: 20

  # 160  Model SN                             20 byte  RW  uint16  ASCII BMS Manufacturer
  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name1} bms serial number"
    skip_updates: 60
    address: 160
    register_type: holding
    register_count: 10
    response_size: 20

  # 170  PACK SN                              20 byte  RW  uint16  ASCII PACK Manufacturer
  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name1} battery pack serial number"
    skip_updates: 60
    address: 170
    register_type: holding
    register_count: 10
    response_size: 20`

JacquesdeBruyn, 2024-07-31:
@syssi I can't seem to get code to post properly so feel free to fix that.
Thanks for all the help along the way.

JacquesdeBruyn, 2024-07-31:
And the cable I made up for this. 
Colour coding is a bit off but the bms has internal bridged connections so you can mirror the left 4 pins with the right 4 pins and have data route through. 

[IMG_20240731_094157_928.jpg](https://github.com/user-attachments/assets/0e57c00f-62c6-41de-a8ab-5480076ac0f4)

JacquesdeBruyn, 2024-08-12:
> I'm confused by these sentences:
> Is this summary correct:
> You have two different battery packs. At the first one the protocol
> PACE_MODBUS is available as inverter protocol in PbmsTools.exe. The
> second battery pack doesn't provide the PACE_MODBUS item at the
> drop-down. You are using SRNE instead. All entities are working?

This first battery is a different brand. It works fine no matter what
protocol it's set to. The Revov works properly when set to SRNE under
rs485. I've had issues with it sending data when the other battery is being
read causing checksum failures. But it seems worse on certain esp boards
and I have to go figure out why that is. For now it's working properly, and
if it doesn't want to play nice with the other battery I'll just put
separate d1 minis for each of them.

syssi, 2024-08-12:
Perfect! I've added your battery bank to the list of supported devices.

closed 2024-08-12 by syssi

JacquesdeBruyn, 2024-08-25:
I cleared my laptop for linux and my backup is at home. But I'll find and post it as soon as ek get back.

nkinnan, 2024-08-29:
@JacquesdeBruyn It's certainly not time-sensitive, so I'm not intending to bug you, just a friendly reminder in case this got forgotten :)  Thanks!

JacquesdeBruyn, 2024-08-29:
@nkinnan  Sorry for the wait, I misplaced my backup drive. Gave it another search when I got home. Anyway, here you go. Be sure to check in if you find something interesting. [Gobel_PC-BMS_Tools.zip](https://github.com/user-attachments/files/16802920/Gobel_PC-BMS_Tools.zip)

nkinnan, 2024-08-29:
No worries, thank you!  I am contributing to the documentation in this repository with what I find, including the results from the software version you shared.  I'm also working an an esphome component for the 0x25 protocol version.