syssi/esphome-pace-bms #54: Strange response from 485

tumbochka, 2024-09-08:
Hello!

We discussed an ability to monitor several battery banks via rs-232 in Seplos project but stucked so I decided to try another approach. I bought a device:

[image]

It has esp32 and rs485.

Then I installed your config using 21 and 22 pins:

```
substitutions:
  name: pace-bms
  device_description: "Monitor and control a PACE BMS via RS485 (Modbus)"
  tx_pin: GPIO22
  rx_pin: GPIO21
  #flow_control_pin: GPIO18

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pace-bms"
    version: 2.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

api:

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    # flow_control_pin: ${flow_control_pin}
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  # - id: bms1
  #   # Slave address 0x02
  #   address: 0x02
  #   modbus_id: modbus0
  #   command_throttle: 200ms
  #   update_interval: 10s    

binary_sensor:
  #  11  Status/Fault flag                     2 byte   R  uint16  Hex See ^3
  #      Bit 8: Status: Charging operation
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} charging operation"
    address: 11
    register_type: holding
    bitmask: 0x0100

  #  11  Status/Fault flag                     2 byte   R  uint16  Hex See ^3
  #      Bit 8: Status: Charging operation
  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name} charging operation"
    address: 11
    register_type: holding
    bitmask: 0x0100

.... and other sensors   
``` 

I changed the 485 protocol to Pace in pbms tools.

And I got the following response:

```
[23:04:56][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[23:04:57][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[23:04:57][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[23:04:57][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[23:04:57][D][uart_debug:114]: >>> 01:03:00:00:00:08:44:0C
[23:04:57][D][modbus_controller:040]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue
[23:04:57][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[23:04:58][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[23:04:58][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[23:04:58][W][modbus:114]: Modbus CRC Check failed! B541!=777
[23:04:58][D][uart_debug:114]: <<< 87:FF:91:77:07:7F:7E:87:FF:91:77:07:7F:7E:87:FF:91:77:07:7F:7E:87:FF:91:77:07:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:7F:7E:FF:FF:EF:77:E7:00:7F:7E:FF:FF:EF:77:E7:00:7F:7E:FF:FF:EF:77:E7:00:7F:7E:FF:FF:EF:77:E7:00:7F:7E:FF:FF:EF:77:E7:00:7F:7E:ED:FF:F7:D7:7A:7F:7E:ED:FF:F7:D7:7A:7F:7E:ED:FF:F7
[23:04:58][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[23:04:58][D][uart_debug:114]: >>> 01:03:00:09:00:04:94:0B
[23:04:58][D][modbus_controller:040]: Modbus command to device=1 register=0x09 countdown=0 no response received - removed from send queue
[23:04:59][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[23:04:59][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[23:04:59][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[23:04:59][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[23:04:59][D][uart_debug:114]: >>> 01:03:00:0F:00:16:F4:07
[23:05:00][D][modbus_controller:040]: Modbus command to device=1 register=0x0F countdown=0 no response received - removed from send queue
[23:05:00][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[23:05:00][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[23:05:00][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[23:05:00][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[23:05:00][D][uart_debug:114]: >>> 01:03:00:3C:00:37:C4:10
[23:05:01][D][modbus_controller:040]: Modbus command to device=1 register=0x3C countdown=0 no response received - removed from send queue
[23:05:04][D][uart_debug:114]: >>> 02:03:00:0B:00:01:F5:FB
[23:05:04][D][uart_debug:114]: >>> 02:03:00:0B:00:01:F5:FB
[23:05:04][D][uart_debug:114]: >>> 02:03:00:0B:00:01:F5:FB
[23:05:04][D][uart_debug:114]: >>> 02:03:00:0B:00:01:F5:FB
[23:05:05][D][uart_debug:114]: >>> 02:03:00:0B:00:01:F5:FB
[23:05:05][D][uart_debug:114]: <<< D7:7A:7F:7E:ED:FF:F7:D7:7A:7F:7E:ED:FF:F7:D7:7A:7F:7E:E1:FF:D3:17:7C:7F:7E:E1:FF:D3:17:7C:7F:7E:E1:FF:D3:17:7C:7F:7E:E1:FF:D3:17:7C:7F:7E:E1:FF:D3:17:7C:7F:7E:87:FF:91:77:07:7F:7E:87:FF:91:77:07:7F:7E:87:FF:91:77:07:7F:7E:87:FF:91:77:07:7F:7E:87:FF:91:77:07:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15:02:BF:FF:E9:FF:FD:15
```

What could I do wrong?

syssi, 2024-09-09:
1. Which port of your BMS do you use for RS485 communication?
2. Please check the DIP switch configuration. Did you select `select address 0x01` here?
3. How does your data cable look like?
4. Did you try to swap GPIO22 and GPIO21 just for testing?

tumbochka, 2024-09-09:
1. The one I used for the solar assistenr previously.

[image]

The 485 that near CAN (the closest one)

2. It's correct, the Master battery.

3. Just cut a network cable and insert orange, orange-white and green-white.

[image]

4. Yep no response at all in this case.

tumbochka, 2024-09-14:
@syssi Any ideas? :)

syssi, 2024-09-14:
Do you have an second ESP and RS485 converter to make some tests?

tumbochka, 2024-09-14:
@syssi  Need to purchase may be :)
Could you please take a look at this thread when you have some time?
https://github.com/syssi/esphome-seplos-bms/discussions/68

tumbochka, 2024-09-20:
@syssi  Hello :)
I bought another ttl & esp32.

[image]

[image]

I attached: orange rs485 to A, orange-white 485 to B, RO to pin16, DI to pin17, vcc and gnd from esp32 to vcc and gnd of the ttl. Is it a correct wiring? Do I need to connect DE and RE somehow? 

Currently the ttl power led lights but it doesn't receive any data.

[image]

syssi, 2024-09-20:
Yes. Your RS485 module requires flow control via RE+DE. See https://github.com/syssi/esphome-solax-x1-mini?tab=readme-ov-file#rs485-ttl-module-with-flow-control-pin

I asked about a second ESP + RS485 converter because I would like to build a test setup to verify your wiring. The test setup should look likes this:

```
ESP32 <--> RS485 Converter <----------> RS485 Converter <--> ESP32
```

We will send some dummy message from the left ESP to the right ESP and vice verse. As soon this is working properly we will attach your BMS.

tumbochka, 2024-09-20:
@syssi  attached RE to flow control and it works now :)  Thank you.
Could the previos esp + ttl have the same issue? It had something like 
[image]
in addition to rx/tx.

syssi, 2024-09-20:
Good catch! It looks like you have to pull some GPIOs high to enable & select the RS485 converter feature.

See https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/example/PlatfromIO/RS485-modbus/src/main.cpp#L28-L35

tumbochka, 2024-09-20:
@syssi  thank you!
One more question: is this solution capable to work with several banks connected?

syssi, 2024-09-20:
Yes. You can talk up to 15 battery banks.

tumbochka, 2024-09-20:
@syssi How can I do it? I tried something like:
```
uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus0
    uart_id: uart_0
    flow_control_pin: ${flow_control_pin}
    send_wait_time: 200ms

modbus_controller:
  - id: bms0
    # Slave address 0x01
    address: 0x01
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s
  - id: bms1
    # Slave address 0x01
    address: 0x02
    modbus_id: modbus0
    command_throttle: 200ms
    update_interval: 10s    

binary_sensor:
  #  11  Status/Fault flag                     2 byte   R  uint16  Hex See ^3
  #      Bit 8: Status: Charging operation
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} charging operation"
    address: 11
    register_type: holding
    bitmask: 0x0100

  - platform: modbus_controller
    modbus_controller_id: bms1
    name: "${name} charging operation --- 2"
    address: 11
    register_type: holding
    bitmask: 0x0100    

```

but it didn't work -- only the first one worked.

syssi, 2024-09-20:
Did you set the dip switches of the second BMS to 0x02? Which RS485 port do you use?

tumbochka, 2024-09-20:
@syssi The switches are set correctly as both inverter (via can) and solar assistant (via rs232) see both packs. The second BMS is set to 0x02. When I connect to the second pack and change the address to 0x02 in yml it works. 

I use RS485 that near CAN.

syssi, 2024-09-21:
It's not enough to connect a single BMS (on the RS485 bus next to the can bus). You have to use multiple data cables all connected to your ESPHome setup.

tumbochka, 2024-09-21:
Do you mean a configuration like this?

[image]

syssi, 2024-09-21:
No. :-)

tumbochka, 2024-09-21:
Thank you!

> Good catch! It looks like you have to pull some GPIOs high to enable & select the RS485 converter feature.
> 
> See https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/example/PlatfromIO/RS485-modbus/src/main.cpp#L28-L35

A question about this change. Will you add this to the solution and config or can I make it myself? I don't see any `cpp` sources in this repo but may be I can find them?

syssi, 2024-09-21:
You have to extend the YAML a bit. I will have look what needs to be done.

syssi, 2024-09-21:
Please extend your YAML configuration by:

```
switch:
  - platform: gpio
    pin: GPIO16
    name: "LilyGO T-CAN485 PIN_5V_EN"
    internal: true
    restore_mode: ALWAYS_ON

  - platform: gpio
    pin: GPIO17
    name: "LilyGO T-CAN485 RS485_EN_PIN"
    internal: true
    restore_mode: ALWAYS_ON

  - platform: gpio
    pin: GPIO19
    name: "LilyGO T-CAN485 RS485_SE_PIN"
    internal: true
    restore_mode: ALWAYS_ON
```

tumbochka, 2024-09-21:
Still doesn't work -- could it be an issue with LilyGO?

```
[21:47:28][W][modbus:114]: Modbus CRC Check failed! 7070!=00
[21:47:29][D][uart_debug:114]: <<< 00:FF:FF:00:00:F8:00
[21:47:29][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:29][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:29][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:29][D][modbus_controller:038]: Modbus command to device=2 register=0x3C no response received - removed from send queue
[21:47:35][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:35][D][uart_debug:114]: <<< 00:00:FE:00:FE:00
[21:47:35][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:35][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:36][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:36][W][modbus:114]: Modbus CRC Check failed! A031!=FF
[21:47:36][D][uart_debug:114]: <<< 00:FE:00:FF:00:FF
[21:47:36][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:36][D][modbus_controller:038]: Modbus command to device=2 register=0x00 no response received - removed from send queue
[21:47:36][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:36][D][uart_debug:114]: <<< 00:00
[21:47:36][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:36][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:37][D][uart_debug:114]: <<< 00:1F
[21:47:37][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:37][D][uart_debug:114]: <<< 00
[21:47:37][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:37][D][uart_debug:114]: <<< 00
[21:47:37][D][modbus_controller:038]: Modbus command to device=2 register=0x09 no response received - removed from send queue
[21:47:37][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:37][D][uart_debug:114]: <<< 00
[21:47:37][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:38][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:38][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:38][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:38][D][modbus_controller:038]: Modbus command to device=2 register=0x0F no response received - removed from send queue
[21:47:38][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:38][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:39][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:39][W][modbus:114]: Modbus CRC Check failed! 3030!=FF
[21:47:39][D][uart_debug:114]: <<< 00:FF:00:FF:00:EF:00:FF:00:00:9F:00
[21:47:39][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:39][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:39][D][uart_debug:114]: <<< 00:00
[21:47:39][D][modbus_controller:038]: Modbus command to device=2 register=0x3C no response received - removed from send queue
[21:47:45][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:45][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:45][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:46][D][uart_debug:114]: <<< 00:00:FF:FF:00
[21:47:46][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:46][D][uart_debug:114]: <<< 00
[21:47:46][D][uart_debug:114]: >>> 02:03:00:00:00:08:44:3F
[21:47:46][D][uart_debug:114]: <<< 00
[21:47:46][D][modbus_controller:038]: Modbus command to device=2 register=0x00 no response received - removed from send queue
[21:47:46][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:46][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:47][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:47][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:47][W][modbus:114]: Modbus CRC Check failed! 3030!=BF00
[21:47:47][D][uart_debug:114]: <<< 00:FF:00:00:BF:00
[21:47:47][D][uart_debug:114]: >>> 02:03:00:09:00:04:94:38
[21:47:47][D][uart_debug:114]: <<< 00
[21:47:47][D][modbus_controller:038]: Modbus command to device=2 register=0x09 no response received - removed from send queue
[21:47:47][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:47][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:48][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:48][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:48][W][modbus:114]: Modbus CRC Check failed! 3030!=FF
[21:47:48][D][uart_debug:114]: <<< 00:FF:00:FF:00:F8:00
[21:47:48][D][uart_debug:114]: >>> 02:03:00:0F:00:16:F4:34
[21:47:48][D][uart_debug:114]: <<< 00
[21:47:48][D][modbus_controller:038]: Modbus command to device=2 register=0x0F no response received - removed from send queue
[21:47:48][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:48][D][uart_debug:114]: <<< 00
[21:47:48][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:49][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:49][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:49][D][uart_debug:114]: >>> 02:03:00:3C:00:37:C4:23
[21:47:49][D][modbus_controller:038]: Modbus command to device=2 register=0x3C no response received - removed from send queue
```

tumbochka, 2024-09-21:
> No. :-)
> 
> ```
>                   RS485                      UART
> ┌────────────┐                   ┌──────────┐                ┌─────────┐
> │            │                   │          │<----- RX ----->│         │
> │    PACE    │<-----B- --+------>│  RS485   │<----- TX ----->│ ESP32/  │
> │    BMS     │<---- A+ --|-+---->│  to TTL  │<----- GND ---->│ ESP8266 │
> │    0x01    │           | |     │  module  │<----- 3.3V --->│         │<-- VCC
> │            │           | |     │          │                │         │<-- GND
> └────────────┘           | |     └──────────┘                └─────────┘
>                          | |
> ┌────────────┐           | |  
> │            │           | |  
> │    PACE    │<-----B- -/  |
> │    BMS     │<---- A+ ----/
> │    0x02    │
> │            │              
> └────────────┘              
> ```
> 
> Your YAML configuraiton above short work ([#54 (comment)](https://github.com/syssi/esphome-pace-bms/issues/54#issuecomment-2364319598)).

I can confirm this works, thank you!

martinarva, 2024-09-27:
Hi! I have the Lilygo T-CAN485 as well and got mine working. Instead of switch, i added this:

output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

Also removed this line from modbus as it was conflicting. So now flow control is always on.

flow_control_pin: ${flow_control_pin}

syssi, 2024-09-27:
Flow control isn't needed for this type of RS485 converter.

closed 2024-09-27 by syssi

tumbochka, 2024-09-29:
> Hi! I have the Lilygo T-CAN485 as well and got mine working. Instead of switch, i added this:
>
> ```
> output:
>   - platform: gpio
>     id: ENABLE_PIN # Enable the chip
>     pin:
>       number: GPIO19 [...]

Confirm this solution works for my LilyGO, thank you!