syssi/esphome-pace-bms #62: Possibility to get connected batteries information from main battery?

martinarva, 2024-12-27:
Hi!

First thank you for what you have done, it's a really useful piece of software.

At the moment i have four batteries and they are connected to one Lilygo RS485 ESP32 device. If i connect to by bms via rs-232 witha Pbms software i can see all four battery packs from the main battery. Is there a possibility to get connected batteries data with this esphome-pace-bms as well instead of connecting to each battery rs485 port? With this i could lose quite a lot cabling.

To my understanding it's somehow solved with this integration: https://github.com/Tertiush/bmspace

syssi, 2024-12-27:
I'm not super deep into the details of the Pace BMS. This is my understanding:

If you talk to the BMS using Pbms via RS232 a Seplos ASCII protocol is used. A frame looks like this:

https://github.com/Tertiush/bmspace/blob/main/pace-bms-dev/bms.py#L460

If you talk to the BMS using this implementation via RS485 the protocol is called Modbus. There are some users around which wasn't successful to setup a RS485/Modbus connection. They are using this project (because of the different protocol):

https://github.com/syssi/esphome-seplos-bms

martinarva, 2024-12-27:
Thanks for a quick reply.

So do I understand correctly that this project is modbus, meaning that i have to connect all batteries to use it and second one is Seplos protocol that uses rs232 connection and i get probably all packs data from the main battery?

syssi, 2024-12-27:
Option 1: Use RS485/Modbus: Chain all RS485 ports next to the CAN bus and attach the Lilygo RS485 to the bus.
Option 2: Use RS232: Attach a ESP32 + RS232 converter to the RS232 port of the master. It looks like your Lilygo CAN485 doesn't have a RS232 converter on board.

martinarva, 2024-12-27:
> Option 1: Use RS485/Modbus: Chain all RS485 ports next to the CAN bus and attach the Lilygo RS485 to the bus. Option 2: Use RS232: Attach a ESP32 + RS232 converter to the RS232 port of the master. It looks like your Lilygo CAN485 doesn't have a RS232 converter on board.

I currently use the option 1 and it actually works great. 
I'm using this approach to optimize the amount of esp32 code: https://github.com/syssi/esphome-pace-bms/pull/59

I just have one issue, that battery 2 voltage is some random number. on discharge it's showing current value as a voltage and other times some random number. If i connect via rs232 and Pbms Tools the value is correct. So for some reason over modbus it has aome bug. Also it seems that time to time the queue is overloaded as update_interval is 10 seconds for all the sensors for all 4 packs. I havent figured out yet how to differentiate the updates so that config entities will be updated less frequently and voltages, amps, balancing status will be updated more frequently.

[image]

syssi, 2024-12-28:
I totally forgot this project as possible solution too: https://github.com/nkinnan/esphome-pace-bms

It's a similar implementation to my `esphome-seplos-bms` project but it's tailored for the PACE BMS (RS232, Seplos like) protocol.

syssi, 2024-12-28:
Your dashboard and setup looks pretty nice. Why do you want to change/touch your setup? :-)

martinarva, 2024-12-28:
> Your dashboard and setup looks pretty nice. Why do you want to change/touch your setup? :-) 

Thanks! Only reason is the issue with battery 2 wrong voltage as I said earlier. 

It's weird that everything else is ok, just one battery pack voltage is showing wrong value. When it's charging it's showing pack current as voltage and other times just a random number. I think that this is some modbus error and thought I try the rs232 and connect only first battery as I can see the correct value with PBms Tools.

But to be honest I'm not sure if it's worth changing the setup

john7doe, 2025-03-31:
@martinarva would you be willing to share your dashboard code? I

closed 2025-07-05 by syssi