syssi/esphome-pace-bms #69: HA Core 2025.5.0 breaks esphome-pace-bms (?)

garylankford8, 2025-05-09:
After updating Core there was no data generated in HA despite the esp32-RS485 being active as evidenced by flashing LEDs, and despite numerous attempts to restart the modules and reinstall the .yaml, no data was generated in HA until Core was regressed to 2025.4.4.

Word of warning...

syssi, 2025-05-09:
Do you use `mqtt` vor the `api` component? Did you see any error messages at the HA log?

garylankford8, 2025-05-09:
api... Nothing current in core logs, and I didn't think to check logs during... my bad, still learning. Logs only go back 12 hrs, and I am past that since the regress. I will try to reproduce the problem by upgrading Core again, but that will have to wait till later today.

syssi, 2025-05-09:
Take your time! :-)

garylankford8, 2025-05-10:
Issue became mute with the release of Core 2025.5.1... I would have had to reinstall 2026.5.0 from a backup and it was easier to just install the latest version, which did not reproduce the problem that I had seen with no data in HA from my SOK batteries.

I consider this issue closed, and if there is ever a next time, I will go straight to logs...

Thanks, syssi!

syssi, 2025-05-10:
You are welcome! Enjoy the sun and your batteries! :-)

closed 2025-05-10 by syssi